raspiblitz/raspiblitz #4575: Feature Request: Electrs Listing in Web GUI for RaspiBlitz only when installed, or change stated info when clicking Electrs

warioishere, 2024-05-16:
**Feature Request: Electrs Listing in Web GUI for RaspiBlitz**

**Issue Description:**
In the current version 1.11 of RaspiBlitz, the Electrs application is listed in the Web GUI on the left located Overview bar. When users click on it, a message appears stating that Electrs needs to synchronize before it can be used. This causes confusion among many new users who assume that Electrs is already installed and they just need to wait for it to synchronize. Consequently, they often wait for a long time and eventually seek help in the RaspiBlitz Telegram chat, asking why Electrs has not finished synchronizing.

**Problem:**
Electrs is not installed by default. It must first be installed through the CLI menu (Menu -> Apps -> Electrum Rust). The current message in the Web GUI does not inform users about this necessity, leading to misunderstandings and unnecessary waiting times.

**Proposed Solution:**
1. **Modify the Electrs Entry in the Web GUI:**
   Update the message displayed when users click on Electrs to clearly state that Electrs must be installed first via the CLI menu. This can help users understand that additional steps are required before synchronization can occur.

   Suggested Message:
   "Electrs is not installed by default. Please install Electrs via the CLI menu: Menu -> Apps -> Electrum Rust."

2. **Conditional Display of Electrs in the Web GUI:**
   Alternatively, adjust the Web GUI so that Electrs is only listed under available apps if it has been installed. This would prevent users from encountering the synchronization message before they have even installed the application.

**Benefits:**
- Reduces user confusion and frustration by providing clear instructions.
- Decreases the number of repetitive queries in the RaspiBlitz Telegram chat.
- Enhances overall user experience by ensuring the Web GUI accurately reflects the system's state.

Thank you for considering this feature request. Implementing these changes will significantly improve the user experience for new RaspiBlitz users.

velgard1, 2024-06-23:
Same issue here, luckily had a look before opening a new issue and trying to figure out if I can fix it myself...

cstenglein, 2024-07-08:
Checked the code for the SideDrawer:
https://github.com/raspiblitz/raspiblitz-web/blob/master/src/layouts/SideDrawer.tsx#L49

It looks if the app is installed, so it shouldn't appear in the sidebar. Looks like the API sends a wrong message.
@rootzoll

rootzoll, 2024-07-18:
Yes there are two other issues open on that - @fusion44 we would need you help here:
- https://github.com/raspiblitz/raspiblitz/issues/4649
- https://github.com/fusion44/blitz_api/issues/259

fusion44, 2024-07-20:
We can fix this easily. See my comment in the other issue. I'm closing this in favor for the other one where the discussion is happening. 

closing in favor for #4649

closed 2024-07-20 by fusion44