raspiblitz/raspiblitz #4585: Can we avoid reboot when changing password B?

rootzoll, 2024-05-24:
Requested on Berlin Bitcoin Meetup.

Have to check if all places where password B is used have a pre-start script that upadtes in local config of the service the passwordB right before start.

And then to ensure that if the user changes password B and we use systemd to restart bitcoind - we need to make sure that all depending apps get also automatically restarted.

So we might need to extend the .service files like:

```
[Unit]
Description=RTL Webinterface
Wants=lnd.service
After=lnd.service
PartOf=bitcoind.service
```

For background:

_The  `PartOf` directive ensures that whenever the lnd.service is restarted, the rtl.service is also restarted._

If RTL gets partof LND or Bitcoind that still can be discussed.

openoms, 2024-05-31:
Related to make restart optional on updates: 
- [ ] https://github.com/raspiblitz/raspiblitz/issues/4585

geco91, 2024-06-04:
Go thru list of services here and either add manual restart or ensure it gets restarted by systemd (because PartOf) when bitcoind gets restarted once all configs were changed.

https://github.com/geco91/raspiblitz/blob/67a0afae8633faa1bbe920e99711e02596dc8089/home.admin/config.scripts/blitz.passwords.sh#L344

Also check which of these services have a `prestart` script already that is auto-setting RPC password in their config before start.

rootzoll, 2024-06-19:
PR merged. SHould be tested during RC phase.

closed 2024-07-08 by rootzoll