raspiblitz/raspiblitz #4613: Error creating persistant SAM session

SepticFrank, 2024-06-04:
2024-06-04T15:21:01Z [i2p] Error listening: Receive timeout (received 0 bytes without terminator before that)
2024-06-04T15:21:01Z [i2p] Creating persistent SAM session 01a3e90a22 with 127.0.0.1:7656
2024-06-04T15:24:01Z [i2p] Error connecting to kh24bjobq56f2njakpigaokokgkdrvve7vuo3m4oh2xm3p2w3lvq.b32.i2p:0: Receive timeout (received 0 bytes without terminator before that)
2024-06-04T15:24:01Z [i2p] Creating persistent SAM session 5f589c6154 with 127.0.0.1:7656

I get these in by bitcoinlog every few minutes. The initial blockchain sync seems to stall sometimes as well more often that not when these pop up. The is the latest SD build and a clean SSD other than syncing the blockchain. 

I'm trying to solve my SSD going into read only as well. I've tried multiple SSDS, power supplys, and new pi4 so I'm trying to rule out every possible reason why.

openoms, 2024-06-04:
If your SSD goes read-only it has likely failed - here are some tests you can run: https://github.com/raspiblitz/raspiblitz/issues/2599

Can also connect it to a PC running Linux (like Ubuntu live) and test it there.

The i2p errors can be ignored, it does not affect your disk.
Can restart the i2pd service with:
```
sudo systemctl restart i2pd
```

SepticFrank, 2024-06-04:
sudo smartctl -a -d sat /dev/sda
smartctl 7.3 2022-02-28 r5338 [aarch64-linux-6.6.20+rpt-rpi-v8] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Marvell based SanDisk SSDs
Device Model:     SanDisk SSD PLUS 1000GB
Serial Number:    191393440108
LU WWN Device Id: 5 001b44 4a888855f
Firmware Version: UH5100RL
User Capacity:    1,000,207,286,272 bytes [1.00 TB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Form Factor:      2.5 inches
TRIM Command:     Available, deterministic
Device is:        In smartctl database 7.3/5319
ATA Version is:   ACS-2 T13/2015-D revision 3
SATA Version is:  SATA 3.2, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Tue Jun  4 16:59:05 2024 BST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00)	Offline data collection activity
					was never started.
					Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0)	The previous self-test routine completed
					without error or no self-test has ever 
					been run.
Total time to complete Offline 
data collection: 		(  120) seconds.
Offline data collection
capabilities: 			 (0x15) SMART execute Offline immediate.
					No Auto Offline data collection support.
					Abort Offline collection upon new
					command.
					No Offline surface scan supported.
					Self-test supported.
					No Conveyance Self-test supported.
					No Selective Self-test supported.
SMART capabilities:            (0x0003)	Saves SMART data before entering
					power-saving mode.
					Supports SMART auto save timer.
Error logging capability:        (0x01)	Error logging supported.
					General Purpose Logging supported.
Short self-test routine 
recommended polling time: 	 (   2) minutes.
Extended self-test routine
recommended polling time: 	 ( 182) minutes.

SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0032   100   100   000    Old_age   Always       -       0
  9 Power_On_Hours          0x0032   100   100   000    Old_age   Always       -       34223
 12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       457
165 Total_Write/Erase_Count 0x0032   100   100   000    Old_age   Always       -       10006
166 Min_W/E_Cycle           0x0032   100   100   ---    Old_age   Always       -       56
167 Min_Bad_Block/Die       0x0032   100   100   ---    Old_age   Always       -       0
168 Maximum_Erase_Cycle     0x0032   100   100   ---    Old_age   Always       -       145
169 Total_Bad_Block         0x0032   100   100   ---    Old_age   Always       -       1520
170 Unknown_Marvell_Attr    0x0032   100   100   ---    Old_age   Always       -       0
171 Program_Fail_Count      0x0032   100   100   000    Old_age   Always       -       0
172 Erase_Fail_Count        0x0032   100   100   000    Old_age   Always       -       0
173 Avg_Write/Erase_Count   0x0032   100   100   000    Old_age   Always       -       56
174 Unexpect_Power_Loss_Ct  0x0032   100   100   000    Old_age   Always       -       449
184 End-to-End_Error        0x0032   100   100   ---    Old_age   Always       -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       131
188 Command_Timeout         0x0032   100   100   ---    Old_age   Always       -       0
194 Temperature_Celsius     0x0022   052   073   000    Old_age   Always       -       48 (Min/Max 13/73)
199 SATA_CRC_Error          0x0032   100   100   ---    Old_age   Always       -       0
230 Perc_Write/Erase_Count  0x0032   100   100   000    Old_age   Always       -       7226 2836 7226
232 Perc_Avail_Resrvd_Space 0x0033   100   100   005    Pre-fail  Always       -       100
233 Total_NAND_Writes_GiB   0x0032   100   100   ---    Old_age   Always       -       58314
234 Perc_Write/Erase_Ct_BC  0x0032   100   100   000    Old_age   Always       -       156347
241 Total_Writes_GiB        0x0030   100   100   000    Old_age   Offline      -       38212
242 Total_Reads_GiB         0x0030   100   100   000    Old_age   Offline      -       621197
244 Thermal_Throttle        0x0032   000   100   ---    Old_age   Always       -       0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
No self-tests have been logged.  [To run self-tests, use: smartctl -t]

Selective Self-tests/Logging not supported

Good? Bad? This is Fresh Setup on previously used SSD. What's strange is I thought SSDs were pretty resilient but I'm still learning...

openoms, 2024-06-04:
Looks like the disk is almost 4 years old with many bad blocks. It is time ti change.

If getting a 2TB disk consider this tested setup https://t.me/RaspiBlitzFAQ/795 : 
>for the 2TB disk I would recommend getting a better power supply loading through the barrel connector on the side.
With that you should have no power issues: https://www.amazon.de/dp/B088M1Q7GF
To be clear this connects to the X825 SATA-USB adapter board inside the metal case and powers the RPi4 as well through the pogo pins so you won't need the USB C power any more, just the barrel connector.

SepticFrank, 2024-06-04:
Ok thanks. What's frustrating is I bought a new SSD and it is the same as the shopping list but a model G27 instead of G26. Brand new and gives me the same bitcoind restarts. But I guess it isn't recommended?: 

The SanDisk SSD Plus 1TB internal SSD models G26 and G27 differ in their interface and form factor. The G26 has a SATA interface and 2.5-inch form factor, while the G27 uses a PCIe interface and M.2 form factor.

SATA interface:

    Provides maximum sequential read speeds of 550 MB/s and write speeds of up to 500 MB/s.
    Uses the older 6 Gb/s SATA III interface.
    Compatible with most desktop and laptop computers that have a SATA port.

PCIe interface:

    Delivers faster maximum sequential read speeds of up to 560 MB/s and write speeds of up to 530 MB/s.
    Uses the newer PCIe Gen 3 x4 interface with four lanes of data transfer.
    Requires an M.2 slot on the motherboard for installation.
    2.5-inch form factor:
    Standard hard drive size that fits into most laptop and desktop drive bays.
    Provides plug-and-play compatibility but is larger in size.
    M.2 form factor:
    Smaller and thinner size that fits into M.2 slots on modern motherboards.
    Allows for greater flexibility in system design and cooling.
    Requires compatible M.2 slot on the motherboard.
    In summary, the G26 offers a SATA interface and 2.5-inch form factor for maximum compatibility, while the G27 uses PCIe and M.2 for faster speeds in a smaller form factor. The choice depends on the interface and form factor supported by the computer.

Sandisk Ssd Plus 1tb Internal Ssd Model G26
The SanDisk SSD Plus 1TB G26 has a SATA interface and 2.5-inch form factor, providing maximum sequential read speeds of 550 MB/s and write speeds of 500 MB/s. It uses the older 6 Gb/s SATA III interface but is compatible with most computers.
Sandisk Ssd Plus 1tb Internal Ssd Model G27
The SanDisk SSD Plus 1TB G27 uses a PCIe interface and M.2 form factor, delivering faster maximum sequential read speeds of 560 MB/s and write speeds of 530 MB/s. It requires an M.2 slot but has a smaller and thinner size than the G26.

Brand new that I bought/causing problems: [https://www.amazon.com/SanDisk-SSD-PLUS-Internal-SDSSDA-1T00-G27/dp/B09NQJMW28/ref=sr_1_1?dib=eyJ2IjoiMSJ9.uNF-mGpAXVVMCp4bvwdf25lZML4hVYPbjm2RlRxfBsGUKV4prxPpRGd58RoFmtPYFsKmfDaSjMYzmiLnMUWlyfZD8r6DoJdtN_9gUiypyRRRYoRk3JkJqN5TCGi3uEM-Tu-tqkVwDmRcvCmGmsVu_AjJ10maD1_trJX74P6ckP0C6H6RVGZ91NPzn_BWUeHgy2CTwmgBnNnEPJE3rqu-eWDtjtYl07ZsxYQy_P9l9dE.lKaZvwd2wxDjLIdXQj4jmBmqbRs7Eye3u8oP3IwOYzw&dib_tag=se&keywords=g27+ssd&qid=1717519743&sr=8-1](url)

Shopping list: [https://www.amazon.com/SanDisk-240GB-Solid-State-SDSSDA-240G-G26/dp/B07D998212/ref=sr_1_2?crid=2RJD6FSNML2LE&dib=eyJ2IjoiMSJ9.AWM11DZ708GFsv-NjRXPRt0WKkqoht63uQ-d90G_0KJgaygmpqysJD3nfiRFQ2VesECghfoA9hAen-IGlYmroxssc8Qt-kDOEjuI1awbjmCJo38z95Nmul6oanKZRuDy_Ihib42W7wskMtgv8JQZwR2IL7-3UCAa2KW3xfraQNeBTcduRQsiTaPElT7bMQdV-nE5UMN-Uzw3kNWcvVFoXkvtTiuxCqzOv2NPozvploc.W6pUmBfQxOS9k7iFTvLZruy56dyrCE-qFLuYHl3bXcw&dib_tag=se&keywords=g26%2Bssd&qid=1717519797&sprefix=g26%2Bssd%2Caps%2C131&sr=8-2&th=1](url)

SepticFrank, 2024-06-10:
So after trying multiple brand new SSDs and a brand new pi 4 I'm pretty sure it was the UGREEN enclosure or the usb cable that attaches itself to the pi4 was bad and causing errors. 

Because removing it and replacing it with the old SATA USB board from the original shopping list I seem to be back to normal. 

Is the SATA USB board not recommended in the long run? I'll close this by the way

closed 2024-06-10 by SepticFrank