raspiblitz/raspiblitz #4615: LIT not opening in browser

thecharacter, 2024-06-06:
I installed LIT, then went to open my browser, but got an error message: Safari cannot connect to the server.

**I tried to do it manually in the command line and got the error:**
lit@raspberrypi:~ $ litcli sessions add --label="default" --type=admin
[litcli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp [::1]:8443: connect: connection refused"

When the Lightning Terminal did come up, it rejected my password, but I know the password is correct because it works with Thunderhub. When I restart the service from the command line, I get the error below.

**Here's what I get from the command prompt:**
litd.service - litd Service
     Loaded: loaded (/etc/systemd/system/litd.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2024-06-06 04:01:10 BST; 34s ago
    Process: 827729 ExecStart=/usr/local/bin/litd (code=exited, status=1/FAILURE)
   Main PID: 827729 (code=exited, status=1/FAILURE)
        CPU: 824ms

**Here's my debug file. It appears something is going on with the TLS handshake:**

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.11.0
chainnetwork: bitcoin / main
 03:47:46 up  2:36,  3 users,  load average: 1.96, 1.89, 1.99

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-06-06 01:13:42 BST; 2h 34min ago
   Main PID: 6819 (bitcoind)
      Tasks: 19 (limit: 4454)
        CPU: 1h 17min 35.845s
     CGroup: /system.slice/bitcoind.service
             └─6819 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Jun 06 01:12:59 raspberrypi systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Jun 06 01:13:42 raspberrypi systemd[1]: Started bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
Jun 06 01:12:59 raspberrypi systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Jun 06 01:13:42 raspberrypi systemd[1]: Started bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2024-06-06T02:09:26Z New outbound-full-relay v1 peer connected: version: 70016, blocks=846685, peer=11687
2024-06-06T02:09:49Z Saw new header hash=00000000000000000001ba78eb5bcdf99c8f32ff22940e06fe91d7921508f7cc height=846686
2024-06-06T02:09:49Z [net] Saw new cmpctblock header hash=00000000000000000001ba78eb5bcdf99c8f32ff22940e06fe91d7921508f7cc peer=74
2024-06-06T02:09:50Z UpdateTip: new best=00000000000000000001ba78eb5bcdf99c8f32ff22940e06fe91d7921508f7cc height=846686 version=0x20200000 log2_work=94.967535 tx=1018988045 date='2024-06-06T02:09:36Z' progress=1.000000 cache=47.6MiB(388440txo)
2024-06-06T02:10:10Z New block-relay-only v1 peer connected: version: 70016, blocks=846686, peer=11756
2024-06-06T02:11:01Z New block-relay-only v1 peer connected: version: 70016, blocks=846686, peer=11842
2024-06-06T02:11:55Z New block-relay-only v1 peer connected: version: 70016, blocks=846686, peer=11916
2024-06-06T02:12:03Z Saw new header hash=0000000000000000000177ddc602fe37a145471da603e22e5e4271f5a6fbaea4 height=846687
2024-06-06T02:12:03Z [net] Saw new cmpctblock header hash=0000000000000000000177ddc602fe37a145471da603e22e5e4271f5a6fbaea4 peer=74
2024-06-06T02:12:04Z UpdateTip: new best=0000000000000000000177ddc602fe37a145471da603e22e5e4271f5a6fbaea4 height=846687 version=0x20c3c000 log2_work=94.967548 tx=1018994194 date='2024-06-06T02:11:42Z' progress=1.000000 cache=47.7MiB(392803txo)
2024-06-06T02:12:25Z [i2p] Error listening: Receive timeout (received 0 bytes without terminator before that)
2024-06-06T02:12:25Z [i2p] Creating persistent SAM session 19c7b0a0a5 with *.*.*.*:7656
[... 488 lines cut ...]
ln_default_channels_total="0"
ln_default_alias=****
ln_default_channels_active="0"
ln_default_wallet_channels_pending="0"
ln_default_channels_pending="0"
btc_default_synced="1"
btc_default_running="1"
btc_default_ready="1"
btc_default_blocks_behind="0"
btc_default_error_full=""
btc_default_blocks_data_kb="640089628"
btc_default_sync_initialblockdownload="0"
btc_default_port="8333"
btc_default_mempool_transactions="139553"
btc_default_sync_percentage="100.00"
btc_default_online="1"
btc_default_activated="1"
btc_default_version="v26.0.0"
btc_default_blocks_verified="846691"
btc_default_error_short=""
btc_default_sync_initial_done="1"
btc_default_peers="39"
btc_default_blocks_headers="846691"
btc_default_sync_progress="0.9999989433130635"
*** POSSIBLE ERROR REPORTS ***

rootzoll, 2024-06-10:
I also had the problem with the password one time - but later it worked. Maybe the first time LND was not ready. Needs further observation.

For next RaspiBlitz v1.11.1 we will update to LIT 0.12.5 .. maybe also this will improve the situation. Please report if you still have problems with LIT once RaspiBlitz v1.11.1 is out.

thecharacter, 2024-06-11:
Will do, Thanks!!!

closed 2024-09-26 by rootzoll