raspiblitz/raspiblitz #4621: [BonusApp] Fedimint

geco91, 2024-06-12:
As from talks with the development team at BTCPrague of https://fedimint.org the fedrated eCash mint is now in a stable state to be deployed to node distribution like RaspiBlitz. As a bonus App caretaker we got recommended https://github.com/Kodylow @Kodylow

The user docs can be found here: https://fedibtc.github.io/fedi-docs/ and there is a good starting tutorial how to setup a federation with setup-leader & setup-followers: https://fedibtc.github.io/fedi-docs/docs/launch-fedimint

geco91, 2024-06-20:
So it makes sense to offer two options side-by-side:
- Fedi-Guardian (create/join)
- Fedi-Lightning Gateway

In talks with the Fedi team there seems two special problems to solve that RaspiBlitz could take care of in the process of setup:

The guardians (or a lightning gateway) needs to be addressable by fixed domain or IP. Tor can be an option for clients addressing a guardian - but especially for the connection between the guardians a more performant connection is needed. So RaspiBlitz could offer an integrated solution like tailscale (or simlar) to build/join a VPN network between guardians.

Also an alternative to Tor for public client connection with domain/IP:port would be best and is a general topic that makes sense to improve options.

rootzoll, 2024-08-30:
PR to add tailscale to RaspiBlitz: #4712

Kodylow, 2024-08-30:
^ once that lands I can work on the fedimint integrations, thanks for the tag!

rootzoll, 2024-09-24:
Hey @Kodylow - we are doing on the 10th of October a RaspiBlitz Hackday in Berlin (one day before the BTC++ eCash event). It would be great if you or somebody of your team can join and exchange/hack with us on adding Fedimint to the RaspiBlitz: https://www.meetup.com/bitcoin-lab-berlin/events/303230291