raspiblitz/raspiblitz #4623: Seems to miss LND.CONF after Migration

velgard1, 2024-06-13:
Long story short, after my node wasn't running well (e.g. freezes & slow) for a while I decided to setup a new one (V1.11) on a 2TB SSD using a migration file. Since I'm using a Pi4 it took some time, but it worked without issues - so I suppose the hardware is ok - but LND is not starting...

I updated LND which did not work first and then tried to compress the channel.db, which resulted in the following error message:
failed to load config: /home/bitcoin/.lnd/lnd.conf:25 malformed key=value ()
Then I had a look in the config file which had an empty line 25 and when I had a closer look realized that the blitz refers to:
mnt\hdd\lnd\lnd.conf

At least to mee it looks like I somehow I have to direct lnd to the right config file or move the file where lnd wants to have it.
Any ideas or more specific information needded in the first plance? Thanks!

velgard1, 2024-06-16:
I totally agree, but sadly I was not able to figure out the reason why it was always running well for a while - e.g. it worked after flashing the SD, so at first I used a new SD but this was not the reason. Reindexing also helped one time, but since it happened again I tried a new SSD - and at least it seems to work fine apart from starting LND now...

Here the LND.CONF - this seems to be the issue for not starting LND because of line 25, I assume that the values have to be rearranged, but I was not able to figure out the exact reason, line 25 is empty...

tor.active=true
tor.v3=true
tor.privatekeypath=/mnt/hdd/lnd/v3_onion_private_key
tor.socks=9050
tor.control=9051
workers.sig=4
workers.write=4
db.bolt.auto-compact=true
db.bolt.auto-compact-min-age=672h
listen=0.0.0.0:9735
restlisten=0.0.0.0:8080
rpclisten=0.0.0.0:10009
bitcoind.rpchost=127.0.0.1:8332
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=abcd1234
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333
bitcoin.node=bitcoind
bitcoin.mainnet=1
bitcoin.active=1
healthcheck.chainbackend.interval=1m30s
healthcheck.chainbackend.timeout=2m0s
healthcheck.chainbackend.attempts=3
rpcmiddleware.enable=true

[healthcheck]

[bitcoin]

[bitcoind]

[bolt]

[workers]

[tor]

[rpcmiddleware]

rootzoll, 2024-06-19:
@velgard1 the `lnd.conf` you posted looks quite unusual. For example all values beginning with `bitcoind.` are normally under the headline`[bitcoind]` and so on ... and on top the headline `[Application Options]` is missing.

velgard1, 2024-06-20:
Great it worked - I inserted [Application Options] as first line and moved everything in the corresponding sections without changing anything else in the order (I left db.bolt in the first section).

closed 2024-06-20 by velgard1