raspiblitz/raspiblitz #4628: [BonusApp] Cashu

geco91, 2024-06-20:
Since a Cashu mint is no longer integrated in LNbits the RaspiBlitz misses the possiblity to run a Cashu mint. The goal here would be to look into options of alternatives to run a mint and und use LND/CL for a lightning gateway.

rootzoll, 2024-10-09:
For adding a new bonus app - a good start is always to copy & customize the bonus-app template:
`/home.admin/config.scripts/bonus.template.sh`

conorhonor111-art, 2026-10-27:
I'd like to help with this issue. Here's my analysis:

**Issue summary:** [BonusApp] Cashu

**My proposed approach:**
1. Reproduce the issue to confirm I understand the expected vs. actual behavior
2. Trace through the relevant code path to identify the root cause
3. Implement a minimal fix that addresses the root cause without unintended side effects
4. Add or update tests to cover the fixed behavior
5. Submit a PR with a clear description of what changed and why

**Before starting:** A few questions that would help scope the work:
- Is there a preferred technical approach or constraint I should be aware of?
- Are there related issues or PRs I should review first?

I can move quickly on this if it's still open — let me know if I should proceed.

conorhonor111-art, 2026-10-27:
I can work on this in `raspiblitz/raspiblitz`: "[BonusApp] Cashu"

**My approach:**
1. Reproduce the issue to confirm I understand expected vs. actual behavior
2. Trace through the relevant code path to find the root cause
3. Implement a minimal fix without unintended side effects
4. Add or update tests to cover the fixed behavior
5. Submit a PR with a clear description of what changed and why

**Before starting:** Any technical constraints I should know? Related PRs or issues to review first? I can move quickly if this is still open.