raspiblitz/raspiblitz #4630: Sync process always cancels.

wilson-ok, 2024-06-22:
Hello, I need some help. 
I have been trying to re-sync a Raspiblitz for a week now. 
I always get stuck at about 8% and keep getting the following message:

[image]

I have followed the hardware recommendation exactly and have already reflashed the SD. I have already tried to reproduce similar errors.
Unfortunately, I don't know what to do and it's not really fun at the moment.

Best regards

Here is the debug report:

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.11.0
chainnetwork: bitcoin / test
 10:39:05 up  3:50,  4 users,  load average: 1.25, 1.21, 1.12

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT                  LOAD   ACTIVE SUB    DESCRIPTION
● bitcoind.service      loaded failed failed Bitcoin daemon on mainnet
● smartmontools.service loaded failed failed Self Monitoring and Reporting Technology (SMART) Daemon

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
2 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
× bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-06-22 07:07:55 BST; 3h 31min ago
   Duration: 17min 13.105s
    Process: 94926 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 94940 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=1/FAILURE)
        CPU: 199ms

Jun 22 07:07:55 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Jun 22 07:07:55 raspberrypi systemd[1]: Failed to start bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
Jun 22 07:07:55 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 5.
Jun 22 07:07:55 raspberrypi systemd[1]: Stopped bitcoind.service - Bitcoin daemon on mainnet.
Jun 22 07:07:55 raspberrypi systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94927]: /bin/chgrp: changing group of '/mnt/hdd/bitcoin': Read-only file system
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94928]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94929]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94930]: grep: /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94931]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94932]: touch: cannot touch '/mnt/hdd/bitcoin/debug.log': Read-only file system
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94933]: chown: changing ownership of '/mnt/hdd/bitcoin/debug.log': Read-only file system
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94934]: chmod: changing permissions of '/mnt/hdd/bitcoin/debug.log': Read-only file system
Jun 22 07:07:55 raspberrypi bitcoind[94940]: Error: Error reading configuration file: specified config file "/mnt/hdd/bitcoin/bitcoin.conf" could not be opened.
Jun 22 07:07:55 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Jun 22 07:07:55 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Jun 22 07:07:55 raspberrypi systemd[1]: Failed to start bitcoind.service - Bitcoin daemon on mainnet.
Jun 22 07:07:55 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 6.
Jun 22 07:07:55 raspberrypi systemd[1]: Stopped bitcoind.service - Bitcoin daemon on mainnet.
[... 232 lines cut ...]
ln_default_sync_progress="44.56"
ln_default_tor="0"
ln_default_sync_chain="0"
ln_default_channels_pending="0"
ln_default_wallet_onchain_pending="0"
btc_default_running="0"
btc_default_sync_percentage="9.13"
btc_default_ready="0"
btc_default_error_full=""
btc_default_error_short=""
btc_default_blocks_behind="463996"
btc_default_sync_initial_done="0"
btc_default_synced="0"
btc_default_blocks_headers="848944"
btc_default_version="v26.0.0"
btc_default_mempool_transactions="0"
btc_default_online="0"
btc_default_sync_progress="0.09126662844208179"
btc_default_blocks_data_kb="71547988"
btc_default_sync_initialblockdownload="1"
btc_default_peers="4"
btc_default_port="8333"
btc_default_activated="1"
btc_default_blocks_verified="384948"
*** POSSIBLE ERROR REPORTS ***

rootzoll, 2024-06-27:
It seems that your filesystem of the SSD slips into a "read-only" state:

```
Jun 22 07:07:55 raspberrypi systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94927]: /bin/chgrp: changing group of '/mnt/hdd/bitcoin': Read-only file system
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94928]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94929]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94930]: grep: /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94931]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94932]: touch: cannot touch '/mnt/hdd/bitcoin/debug.log': Read-only file system
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94933]: chown: changing ownership of '/mnt/hdd/bitcoin/debug.log': Read-only file system
Jun 22 07:07:55 raspberrypi bitcoin.check.sh[94934]: chmod: changing permissions of '/mnt/hdd/bitcoin/debug.log': Read-only file system
Jun 22 07:07:55 raspberrypi bitcoind[94940]: Error: Error reading configuration file: specified config file "/mnt/hdd/bitcoin/bitcoin.conf" could not be opened.
```

Why this is - not clear .. 

"Smartctl open device: /dev/sda failed: No such device" but df says its mounted
```
*** MOUNTED DRIVES ***
df -T -h
Filesystem Type Size Used Avail Use% Mounted on
/dev/sda1 ext4 916G 78G 793G 9% /mnt/hdd
...
```

@openoms have you seen such error in the past?

openoms, 2024-06-28:
Yes, it is a better form of SSD failure.

Please send another:
`debug -l`
report link after restart or to see how long the SSD was used.
If the disk is new it's likely a power failure which could be solves with using an X825 SATA adapter board with the barrel connector. That can drive the 2TB disks as well.

wilson-ok, 2024-06-28:
Hello

The error seems to have been found.
I have replaced another SSD housing and since then no more crashes. However, I had used the enclosure from the hardware recommendation. I had the problems with this one.

Thanks for the help.

closed 2024-09-23 by cstenglein