raspiblitz/raspiblitz #4633: Lightning Progress stuck after update. Blockchain Progress varies between 100 and 99.99%

rootzoll, 2024-06-27:
### Discussed in https://github.com/raspiblitz/raspiblitz/discussions/4624

<div type='discussions-op-text'>

<sup>Originally posted by **ivaylogochev** June 16, 2024</sup>
Hello!
After few months offline I have tried to run the node again, but it was not responding so I have updated it to 1.10.0.
The peers started to connect and the Blockchain Progress get to grow.  At point it reached 100% and Lightning Progress was 94.**% and it was not growing (the lightning). I have rebooted the LN progress went to 93.57% and I saw the Blockchain Progress varies between 99.99 and 100%

[image]
[image]
[image]

https://[l.termbin.com/f9pi](https://l.termbin.com/f9pi)
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / main
 19:38:05 up 1 day, 20:51,  2 users,  load average: 2.30, 2.63, 2.62

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2024-06-14 23:57:53 BST; 1 day 19h ago
    Process: 6293 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 6430 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 6494 (bitcoind)
      Tasks: 18 (limit: 4531)
        CPU: 6h 58min 43.958s
     CGroup: /system.slice/bitcoind.service
             └─6494 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Warning: journal has been rotated since unit was started, output may be incomplete.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Sat 2024-06-15 01:20:33 BST, ends at Sun 2024-06-16 19:38:05 BST. --
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:16Z WARNING: request rejected because http work queue depth exceeded, it can be increased with the -rpcworkqueue= setting
2024-06-16T18:06:51Z UpdateTip: new best=0000000000000000000291419d679f62a4bf8f787ab3875fe291d6406f67b083 height=848217 version=0x200a8000 log2_work=94.987899 tx=1025051261 date='2024-06-16T17:55:09Z' progress=0.999997 cache=214.0MiB(1478016txo)
2024-06-16T18:06:51Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2024-06-16T18:06:51Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2024-06-16T18:06:52Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2024-06-16T18:06:52Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2024-06-16T18:10:36Z UpdateTip: new best=000000000000000000035c9017501cc010e3892ba79a898c9fe65c3f5b1ea46b height=848218 version=0x30d22000 log2_work=94.987912 tx=1025052564 date='2024-06-16T17:58:15Z' progress=0.999997 cache=214.8MiB(1484247txo)
2024-06-16T18:11:20Z New outbound peer connected: version: 70016, blocks=848218, peer=1873 (outbound-full-relay)
2024-06-16T18:11:57Z Saw new header hash=00000000000000000002b9288f841a4ba58bd57ebeaa4b92f2dd68f40377b8c3 height=848219
2024-06-16T18:11:57Z [net] Saw new cmpctblock header hash=00000000000000000002b9288f841a4ba58bd57ebeaa4b92f2dd68f40377b8c3 peer=1815
2024-06-16T18:12:15Z Socks5() connect to ***.onion:8333 failed: host unreachable
2024-06-16T18:15:14Z UpdateTip: new best=00000000000000000002b9288f841a4ba58bd57ebeaa4b92f2dd68f40377b8c3 height=848219 version=0x2a172000 log2_work=94.987925 tx=1025055614 date='2024-06-16T18:11:47Z' progress=0.999999 cache=216.0MiB(1492950txo)
2024-06-16T18:15:14Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2024-06-16T18:15:14Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2024-06-16T18:15:14Z BlockUntilSyncedToCurrentChain: txindex is </div>

Whickey344, 2024-07-05:
Im new to this so im not sure on what to do

rootzoll, 2024-10-01:
Cling because of duplicate - see #4684

closed 2024-10-01 by rootzoll