raspiblitz/raspiblitz #4634: about router with VPN for raspblitz?

up1son, 2024-06-29:
Does anyone know or can give feedback about putting a modem router with VPN for raspblitz and if it works? because if you look at the electrum service part it shows my local IP, I would like to make it private with this VPN router

stonedbear, 2024-07-01:
why not use wireguard?

here is a good guide that will get you started: https://github.com/TrezorHannes/Dual-LND-Wireguard-VPS

up1son, 2024-07-01:
> why not use wireguard?
> 
> here is a good guide that will get you started: https://github.com/TrezorHannes/Dual-LND-Wireguard-VPS

In this case, the link you sent is to a raspblitz that has a Lightning Node. My situation is just a node, I don't have LN, is it possible to do it via wireguard for just one bitcoin node? I didn't know about this wireguard tool.

stonedbear, 2024-07-02:
wireguard is pretty versatile. In simplest terms you can create an encrypted tunnel between 2 or more computers so they can communicate like over local network. So totally doable. 

The guide above goes one step further and allows you to expose your node on a public ip without revealing your actual ip where node runs from.