raspiblitz/raspiblitz #4646: Cannot restore with seed + scb / no new password a

dragsugg, 2024-07-11:
My ssd crashed and i wanted to restore with a new ssd with seed and scb. i cannot type in a new password a. it crashes and showes the following error:

###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.11.1rc1
Thu 11 Jul 22:02:55 BST 2024
***********************************************
  UNIT                      LOAD   ACTIVE SUB     DESCRIPTION
  avahi-daemon.service      loaded active running Avahi mDNS/DNS-SD Stack
  background.service        loaded active running RaspiBlitz Background Monitoring Service
  cron.service              loaded active running Regular background program processing daemon
  dbus.service              loaded active running D-Bus System Message Bus
  fail2ban.service          loaded active running Fail2Ban Service
  NetworkManager.service    loaded active running Network Manager
  polkit.service            loaded active running Authorization Manager
  rsyslog.service           loaded active running System Logging Service
  smartmontools.service     loaded active running Self Monitoring and Reporting Technology (SMART) Daemon
  systemd-fsckd.service     loaded active running File System Check Daemon to report status
  systemd-journald.service  loaded active running Journal Service
  systemd-logind.service    loaded active running User Login Management
  systemd-rfkill.service    loaded active running Load/Save RF Kill Switch Status
  systemd-timesyncd.service loaded active running Network Time Synchronization
  systemd-udevd.service     loaded active running Rule-based Manager for Device Events and Files
  triggerhappy.service      loaded active running triggerhappy global hotkey daemon
  vnstat.service            loaded active running vnStat network traffic monitor

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
17 loaded units listed.
# make sure SSH server is configured & running
# *** /home/admin/config.scripts/blitz.ssh.sh checkrepair
# countKeyFiles(6)
# OK: SSHD RUNNING
## prepare raspiblitz temp
## INIT raspiblitz.info
state=starting
message=starting
setupPhase=boot
setupStep=0
baseimage=raspios_arm64
cpu=aarch64
blitzapi=off
displayClass=headless
displayType=
fsexpanded=0
btc_mainnet_sync_initial_done=0
btc_testnet_sync_initial_done=0
btc_signet_sync_initial_done=0
ln_lnd_mainnet_sync_initial_done=0
ln_lnd_testnet_sync_initial_done=0
ln_lnd_signet_sync_initial_done=0
ln_cl_mainnet_sync_initial_done=0
ln_cl_testnet_sync_initial_done=0
ln_cl_signet_sync_initial_done=0
# raspi_bootdir(/boot/firmware)
## redisEnabled(1)
## INIT RaspiBlitz Cache ... wait background.scan.service to finish first scan loop
INFO: _bootstrap.scan.sh loop started > sudo journalctl -f -u background.scan
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime(2)
No Wifi config by file on sd card.
OK - logs are at 25 MB - within safety limit
Waiting for HDD/SSD ...
total 624
drwxr-xr-x   4 root root   4096 Jan  1  1970 .
drwxr-xr-x 135 root root  12288 Jul 11 22:02 ..
[... 149 lines cut ...]
sda
└─sda1      ext4   BLOCKCHAIN
mmcblk0
├─mmcblk0p1 vfat   bootfs
└─mmcblk0p2 ext4   rootfs
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
# migrationFile()
# Sourcing /mnt/hdd/raspiblitz.conf
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.11.1rc1'
lcdrotate='1'
lightning=''
network='bitcoin'
chain='main'
hostname=''
runBehindTor='on'
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
##################
ERROR in _bootstrap.sh --> missing-passworda on Thu 11 Jul 22:07:12 BST 2024\nDETAIL --> \nDEBUG --> /home/admin/raspiblitz.log
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(missing-passworda)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off

rootzoll, 2024-07-18:
So I was setting up a test scenario first with a fresh node with LND to store "Seedwords & SCB file". Then with v1.11.1rc1 was doing another setup of a fresh node while using the "Seedwords & SCB file" to restore the LND node.

@dragsugg it ran thru without problems. So the error you had seems to be some edge case that might be hardware related.

dragsugg, 2024-07-18:
Ok, so you were able to assign a new password A? Strange.

rootzoll, 2024-07-18:
If somebody runs again into this error, please help to debug one step further and check if the file with the setup data exists and has valid data - try the following two commands on the terminal and post the results:

`sudo ls /var/cache/raspiblitz/temp/raspiblitz.setup`
`sudo cat /var/cache/raspiblitz/temp/raspiblitz.setup`

Because the passwordA should be in this source file - so either the file is missing or the data in it is false formatted:
https://github.com/raspiblitz/raspiblitz/blob/ad5ed24e7322419c89005c6ba6e04a1667974905/home.admin/_bootstrap.sh#L876-L877