raspiblitz/raspiblitz #4648: Check Tor keys & update

rootzoll, 2024-07-17:
Tor Key expired, need to add the new one
They also have an APT repo for the Keys, so we dont need to add them manually
And Tor 0.4.8 has the PoW protection, which we could enable

cstenglein, 2024-07-27:
More info:

When installing tor currently, the installation fails:

```
W: Failed to fetch tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org/dists/bookworm/InRelease  The following signatures were invalid: EXPKEYSIG 74A941BA219EC810 deb.torproject.org archive signing key
```

We could use the package `apt install apt-transport-tor` for the keys:

https://support.torproject.org/apt/apt-over-tor/

Alternative:

```
wget -qO- https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --dearmor | tee /usr/share/keyrings/deb.torproject.org-keyring.gpg >/dev/null
```

See: https://support.torproject.org/apt/tor-deb-repo/

rootzoll, 2024-07-29:
So it the moment we have tor installed on the base image and we should be good for v1.11.1 & 1.11.2:

```
admin@192.168.178.60:~ ₿ tor --version
Tor version 0.4.7.16.
Tor is running on Linux with Libevent 2.1.12-stable, OpenSSL 3.0.13, Zlib 1.2.13, Liblzma 5.4.1, Libzstd 1.5.4 and Glibc 2.36 as libc.
Tor compiled with GCC version 12.2.0
```

But yes for the next build of the base image with v1.12.0 we should adapt the install. Moving to v1.12.0 milestone

openoms, 2024-08-10:
In the meantime:

to refresh the Tor key use:

menu - UPDATE - TOR

or run:
```
wget -qO- https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/torproject.gpg >/dev/null
```
https://t.me/RaspiBlitzFAQ/800

rootzoll, 2024-08-15:
For v1.11.2 & v1.11.3 this will be updates with PR #4697 adding the new key by release script.

Moving issue to v1.12.0 with TODOS:
- [ ] Check if new base image is still oudated signing key
- [ ] Then move updated key to sd build script or just delete from release script:

https://github.com/raspiblitz/raspiblitz/blob/0dca25aa91f95c74ed96b4836f75eba44af4e03a/home.admin/config.scripts/blitz.preparerelease.sh#L68

rootzoll, 2025-05-18:
OK this looks good now for v1.12.0 - closing.

closed 2025-05-18 by rootzoll