raspiblitz/raspiblitz #4649: After fresh Setup Eletrs is already installed

rootzoll, 2024-07-18:
used a v1.11.1rc1-fatpack to setup a new node - when done under WebUI dashboard Eletrs was listed as already installed.

[image]

But its not in the config:

```
admin@192.168.178.60:~ ₿ cat /mnt/hdd/raspiblitz.conf
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.11.1rc1'
lcdrotate='1'
lightning='lnd'
network='bitcoin'
chain='main'
hostname='Testblitz'
runBehindTor=on
mainnet=on
blitzapi=on
lnd=on
setnetworkname=0
i2pd=on
displayClass='lcd'
notifyMethod='mail'
notifyMailTo='mail@example.com'
notifyMailServer='mail.example.com'
notifyMailPort='587'
notifyMailHostname='Testblitz'
notifyMailFromAddress='rb@example.com'
notifyMailFromName='RB User'
notifyMailUser='username'
notifyMailPass='password'
notifyMailEncrypt=off
notifyMailToCert='/mnt/hdd/notify_mail_cert.pem'
notifyExtCmd='/usr/bin/printf'
```

and asking the electrs script on its status it also says its not installed:
```
admin@192.168.178.60:~ ₿ ./config.scripts/bonus.electrs.sh status
##### STATUS ELECTRS SERVICE
version='v0.10.4'
configured=0
infoSync='Service not installed'
installed=1
serviceRunning=0
```

So it seems that its false reported by the API to the WebUI .. @fusion44 how does the API determine that Electrs is installed or where is this coming from?

fusion44, 2024-07-20:
The fix is easy, we just have to determine how to fix it. The problem is that the electrs script returns `installed=1` event when electrs is technically not installed as it is not configured. Every other script doesn't make the distinction between `configured` and `installed`. I'd prefer to fix this on script side by just ending the script early and returning `installed=0`

```bash
if [ "${ElectRS}" = "on" ]; then
  echo "configured=1"
else
  echo "configured=0"
  echo "infoSync='Service not installed'"
  echo "installed=0"
  exit
fi
```

I've changed my bonus.electrs.sh script like this and it worked without adding a special case to the API code.

Although I don't know what this code does:
```bash
if id "electrs" &>/dev/null; then
  echo "installed=1"
else
  echo "installed=0"
fi
```
Is it necessary to run even when the electrs service is off?

rootzoll, 2024-07-20:
> To clarify, other scripts seem to make the distinction between configured and installed but they do return installed=0 when the service is off.

Yeah normally `installed` should be signaling that its pre-installed like with fatpack. And only if `configured` its really activated. Lets keep it for the v1.11.x the old (not correct way) and I will change the electrs script for it ... and I created for v1.12.0 then issue #4650 to change that for all scripts and then the new versioned API for v1.12.x can adapt to that change.

fusion44, 2024-07-20:
I could also adapt the API to just use for `configured` and ignore `installed` completely. This is somewhat confusing  though.

rootzoll, 2024-07-20:
OK I merged #4651 .. this is OK for now.

> I could also adapt the API to just use for configured and ignore installed completely. This is somewhat confusing though.

That should be fine too and already reflect the changes for the on #4650 for the future.

Sorry for the bad naming ... maybe `fatpacked` & `installed` would have been better status names.

closed 2024-08-05 by rootzoll