raspiblitz/raspiblitz #4650: [Scripts/API] Correctly use installed/configured

rootzoll, 2024-07-20:
See https://github.com/raspiblitz/raspiblitz/issues/4649#issuecomment-2241295258 .. with v1.12.0 we should correct all fatpack scripts to correctly show `installed` and `configured` status and the API then need to adapt to that change for the new version.

fusion44, 2024-07-20:
Ok, as I see it right now:
- `configured` determines whether the service is actually *running* (status `on` or `off`)
  - this is switched by clicking `install` on the UI
- `installed` determines whether the *binaries* are installed on the system via some means
  - this is irrelevant for the webui

fusion44, 2024-07-21:
Okay sorry. Now that I understand the problem better it is an API problem. I always mistook `installed` as `on`. The distinction makes sense in a RaspiBlitz context. 

I'll fix it properly. Longer term, maybe we can rename the `installed` var to something like `binary_exists` or `binary_installed` to make it more clear for other new contributors.