raspiblitz/raspiblitz #4667: Numerous 'Lightning Wallet Locked'

thephizzz, 2024-07-29:
**Describe the bug**
With a running node, I receive a 'Lightning Wallet Locked' message every few days.
I unlock the wallet; the node syncs and runs fine again.
This has been going on for a month. It is more of an annoyance than a real problem, but any insight would be appreciated.
Thank you.
Jim

**To Reproduce**
Just start the node and wait. Unlock lightning wallet when asked.

**Debug file**
https://termbin.com/5rrs

**system
Raspiblitz v1.11.0
bitcoin v26.0.0
lnd 0.17.3-beta

rootzoll, 2024-07-29:
In general a request for an unlock happens when LND restarts. So the question is why is your LND restarting. Cannot see anything would in LND logs - just requests to unlock.

The release of RaspiBlitz v1.11.1 is close - best would be if you can so the update once its out and if the problem pesists write again here.

cstenglein, 2024-07-29:
What Bugs me ist this

> error creating wallet config: unable to set up wallet password listeners: shutting down

See also https://github.com/lightningnetwork/lnd/issues/3612

Ist your Swap enabled?

Check with `cat /proc/swaps`