raspiblitz/raspiblitz #4668: Node keeps going offline, no idea why

Delivator, 2024-07-29:
**Describe the bug**
Pretty simple, my node goes offline and I can't open any apps, webui or ssh. Can't even ping it. I had it running with wifi for a while so I thought wifi instability caused it but now I switched to LAN and it still happens. Sometimes I can ping and connect to it again after a while but right now it has been offline for more than 12 hours. Very annoying because I can't figure out why and it's just not reliable. And every time it has been offline for a bit it needs to resync which takes aaaages even if its just a few block missing.

**To Reproduce**
Plug node into wall and lan switch, try to connect.

**Debug file**

Will be provided when it's online again.

Delivator, 2024-07-29:
Node came online again after a restart and it seems like it's using both WiFi and LAN right now. So I assue it could still be a WiFi issue but I have no idea why it stopped working even when LAN was still plugged in.

rootzoll, 2024-07-29:
Please exit to terminal from main menu and run command `./config.scripts/internet.wifi.sh off` and then restart. See if problem still exists.

Delivator, 2024-07-29:
Thank you for the reply! I followed a different guide to disable wifi completely via an option in the `/boot/firmware/config.txt` file. Seems to be working fine now! Would it be better to use the script you mentioned or should the option I took be fine?

rootzoll, 2024-08-01:
I think your option should work fine. Closing issue.

closed 2024-08-01 by rootzoll