raspiblitz/raspiblitz #4677: On Debian 12.6 VM, `build_sdcard.sh` always fails after cutting off Internet on its own

woiwoiwoiwoi, 2024-08-05:
**Describe the bug**
The `build_sdcard.sh` script always seems to cut off Internet connection at a specific moment. After a reboot, the script then completes successfully. As far as I can tell, the last package that is installed and configured is `recolvconf`.

**To Reproduce**
Using virt-manager, I installed Debian, using the `debian-12.6.0-amd64-DVD-1.iso` image. I also tried out the OSBoxes 64 Bit 12.6 image. The `build_sdcard.sh` script from the dev channel always aborts here:

[image]

After a reboot, the script completes successfully and everything appears to be working fine.

tbshfr, 2024-08-18:
I can confirm the bug, after the package resolvconf is installed, dns resolving does not work anymore and the script exits because deb.debian.org cant be resolved.
After a reboot the install works.

rootzoll, 2024-08-18:
Please give is two weeks .. VM install on debian is the next big TODO on our list.

rofra, 2024-08-29:
I wrote a pull request #4715 for solving the issue

closed 2024-11-07 by rootzoll