raspiblitz/raspiblitz #4678: Add Bitcoin Knots as an additonal app (to replace core)

rootzoll, 2024-08-05:
As requested some people want to see Bitcoin Knots on the RaspiBlitz:
https://x.com/ChrisMartl/status/1820527897028989031
https://x.com/LukeDashjr/status/1820565013725372860

The "dont ask, just do it" way would be to fork RaspiBlitz (KnotsBlitz would be my recommended name) and keep it in sync with RaspiBlitz while only taking care of always replacing Bitcoin Core with Knots in the Bitcoin install scripts:
https://github.com/raspiblitz/raspiblitz/blob/dev/home.admin/config.scripts/bitcoin.install.sh
https://github.com/raspiblitz/raspiblitz/blob/dev/home.admin/config.scripts/bitcoin.update.sh

The more advanced integration would be to add "Bitcoin Knots" as an additional app that will deactivate Bitcoin Core and install/activate Knots as a in-place replacement.

_In both solutions make sure that all Ports & CLI stays the same with Knots as it with Core._

This issue is tagged with `help wanted` because at the moment there is no much capacity within main raspiblitz team to take the lead on this feature request - but if someone like to contribute we are happy to give directions and assist.

Retropex, 2024-08-06:
Taking care of this one 👍

https://github.com/Retropex/knotsblitz

rootzoll, 2024-08-06:
@Retropex We like to support forks of RaspiBlitz .. if there is anything that makes it easier for you to maintain/keepInSync the KnotsBlitz - let us know

Retropex, 2024-08-10:
@rootzoll may I ask you if local build worked fine lately, it's probably a misconfiguration on my side but I don't know what to do to fix this:

```
# Build the image
# Running: packer build  -var branch=knots -var pack=fatpack -var github_user=Retropex build.arm64-rpi.pkr.hcl
???????: Failed to load source type

  on build.arm64-rpi.pkr.hcl line 8:
  (source code not available)

Unknown address type: 0

arm.raspiblitz-arm64-rpi: output will be in this color.

==> Wait completed after 120 microseconds

==> Builds finished but no artifacts were created.
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0xbc1188]

goroutine 114 [running]:
github.com/hashicorp/packer/packer.(*CoreBuild).Run(0xc0006e5560, {0x44371e0, 0xc0000b5380}, {0x443c960?, 0xc000546810})
	github.com/hashicorp/packer/packer/build.go:225 +0x5e8
github.com/hashicorp/packer/command.(*BuildCommand).RunContext.func1()
	github.com/hashicorp/packer/command/build.go:255 +0x1da
created by github.com/hashicorp/packer/command.(*BuildCommand).RunContext
	github.com/hashicorp/packer/command/build.go:246 +0xac5

!!!!!!!!!!!!!!!!!!!!!!!!!!! PACKER CRASH !!!!!!!!!!!!!!!!!!!!!!!!!!!!

Packer crashed! This is always indicative of a bug within Packer.
A crash log has been placed at "crash.log" relative to your current
working directory. It would be immensely helpful if you could please
report the crash with Packer[1] so that we can fix this.

[1]: https://github.com/hashicorp/packer/issues

!!!!!!!!!!!!!!!!!!!!!!!!!!! PACKER CRASH !!!!!!!!!!!!!!!!!!!!!!!!!!!!
make: *** [Makefile:108 : arm64-rpi-fatpack-image] Erreur 1
```

rootzoll, 2024-08-13:
@openoms can you comment on the packer build?

@Retropex I would recommend building manual for testing & releases - see FAQ here: https://docs.raspiblitz.org/docs/faq/dev#what-is-the-process-of-creating-a-new-raspberrypi-sd-card-image-release-manually

openoms, 2024-08-14:
@Retropex see the changes here to fix the local (non-docker) packer run: https://github.com/raspiblitz/raspiblitz/pull/4696/files

PR #4968 (2025-05-14): add knots option
- [x] That its based against the `dev` branch - not the default release branch.

This is not ready to merge, its to start receiving advice on what should I add/remove.

Close #4678
closed 2025-05-14 by rootzoll