raspiblitz/raspiblitz #4679: input/output error

pitower, 2024-08-06:
Hey folks, I keep getting input/output errors on my Raspiblitz, which then leads to the well known 6 restarts message. Obviously the HDD can no longer be accessed. I use a 2 TB SSD hard disk Raspiblitz V1.11. After reboots, hard disk access is regularly possible again for some time. This is especially not nice when ready btw on-chain. I have created a debug.log, after a recent reboot. Do you know a solution for this issue?
[debug.log](https://github.com/user-attachments/files/16507131/debug.log)

rootzoll, 2024-08-06:
This looks bad. You have "smartmontools.service" as a failed service which is a standard system service to monitor the health of your SSD and when this has problems to run, it looks like a hardware problem.

```
*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT                  LOAD   ACTIVE SUB    DESCRIPTION
● smartmontools.service loaded failed failed Self Monitoring and Reporting Technology (SMART) Daemon

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
1 loaded units listed.
```
Lots of 2TB SSDs (external over USB on RaspberryPi4) had too high power consumption and were not stable. Is this your setup?

pitower, 2024-08-06:
Yes, this is exactly my setup. I am just trying to sync the 2 TB SDD with a new USB-HUB, which provides an own Power-Adapter.

pitower, 2024-08-06:
Do not work, like intended, does this mean 2 TB SSD can not be deployed with raspiblitz?

pitower, 2024-08-06:
should I quit to deploy a 2TB-SSD?

openoms, 2024-08-10:
Please record the exact power supply, USB adapter and SSD model used.

I have no issues with 2 TB NVMe drives tried so far.

cstenglein, 2024-09-30:
@pitower do you have more information for openoms so we can help you to resolve this?

pitower, 2024-10-01:
> @pitower do you have more information for openoms so we can help you to resolve this?

I have bought a more expensive Nano-Flash SSD with SLC (single-level-cell) Storage. That has done it for me.

closed 2024-10-01 by rootzoll