raspiblitz/raspiblitz #4681: RPI5: NVMe not detected after 1.11.1 Upgrade

tysecure, 2024-08-06:
**Describe the bug**
Hardware used:
• RPI5 8GB
• GeeekPi N04 M.2 NVMe to PCIe Adapter for Raspberry Pi 5 (https://www.amazon.com/dp/B0CRK4YB4C?psc=1&ref=ppx_yo2ov_dt_b_product_details)
• WD_BLACK 2TB SN850X NVMe (https://www.amazon.com/dp/B0B7CMZ3QH?ref=ppx_yo2ov_dt_b_product_details&th=1)

Note: Everything was working great with 1.11.0. I was booting from my microSD card.

After the upgrade, the NVMe is not detected and hence, Raspiblitz is asking me to start fresh. If I attempt to do so, it naturally fails because it cannot detect the NVMe drive. 

I would like to resolve this in order to preserve the data on the drive and carry on with the upgrade vs. starting over.

**To Reproduce**
Steps to reproduce the behavior.

**Debug file**

If possible, please add a debug report. To do that, drag a text file with the debug report in here.
[log.txt](https://github.com/user-attachments/files/16509083/log.txt)

rootzoll, 2024-08-06:
The log.txt is the logs while you already try a fresh setup - when ssh in to a fresh sd card image can you CANCEL the setup menu to get to terminal and use the command `debug` - see details here: https://docs.raspiblitz.org/docs/faq#debug

Also if you flash an older v1.11.0 image and start with that - does it stil lwork?

tysecure, 2024-08-06:
Unfortunately, I'm unable to test this as I'm traveling. However, I suspect it won't fix the issue since the NVMe detection config is no longer on the SD card, especially in version 1.11.0.

usekevin1988, 2024-10-03:
i had the same problem with this board. I bought another one (Geekworm X1002 ) and that will do the job fine

rootzoll, 2024-10-03:
There are no recent changes in the dev branch of the build_sdcard.sh file that directly affect NVMe drive support since v1.11.0 .. so hard to say where this is coming from.

@tysecure can you try again with a v1.10.0 sd card and report if there gets detected again?