raspiblitz/raspiblitz #4682: [1.11.1/dev] build_sdcard.sh failing

adevoss, 2024-08-06:
Script fails:
⠙ Generating browser application bundles (phase: sealing)...Killed
FAIL - npm run build did not run correctly, aborting (1)

**To Reproduce**
- install Debian 12 (no GUI) on a VM.
- execute `build_sdcard.sh -b dev -d headless -t 0 -w off` from `dev` branch

adevoss, 2024-08-18:
`build_sdcard.sh -b dev -f 0 -d headless -t 0 -w off` works (no fatpack, now v1.11.2)

rootzoll, 2024-08-18:
Give us 2 weeks - to optimze the VM install is next after v1.11.2 release

rootzoll, 2024-09-26:
@adevoss I did some successfull build on Debian12 for VMs - if you find the time please retry with latest `build_sdcard.sh` from `dev` branch - thx.

adevoss, 2024-09-29:
`build_sdcard.sh -b dev -f 0 -d headless -t 0 -w off` works. Build and Recovery.  (displays v1.11.2)
But still #4568

rootzoll, 2024-09-30:
Thanks - so closing this issue. #4568 is still on my list for v1.11.3

closed 2024-09-30 by rootzoll