raspiblitz/raspiblitz #4699: [WebUI/API] CLN seedwords dont get listed correctly with numbers

rootzoll, 2024-08-15:
When installing a fresh Core Lightning Wallet thru the redesigned Setup WebUI the seed words looks like this:

[image]

Not sure if this is to fix thru WebUI @cstenglein or API @fusion44 or even the wallet creating shell script @openoms .. but the data structure should be similar to the LND so that it gets rendered correctly. 

Its not critical but should get fixed for 1.11.3 release.

cstenglein, 2024-08-15:
The seedwords are expected in this format:

https://github.com/raspiblitz/raspiblitz-web/blob/master/backend-mock/setup.js#L125

They are split by ", " https://github.com/raspiblitz/raspiblitz-web/blob/master/src/pages/Setup/FinalDialog.tsx#L24

Most likely it is because they are separated by just an empty space, not a comma and an empty space.

rootzoll, 2024-09-23:
@rootzoll check script for same splitting char as LND

cstenglein, 2024-09-26:
If splitting is not done in the LND script, I'll handle the case in the webUI

rootzoll, 2024-09-26:
OK comitted to `dev` branch - cln script should now deliver thru API the comma separated list - like in this example:
```
seedwords='wall,input,chronic,gaze,tip,explain,fog,receive,kingdom,they,corn,universe,soup,roast,remind,country,make,harsh,dream,cross,all,magic,napkin,found'
```

rootzoll, 2024-09-26:
TODO-TEST: With v1.11.3rc1 test fresh setup of cln node and check seed word listing.

cstenglein, 2024-09-26:
> OK comitted to `dev` branch - cln script should now deliver thru API the comma separated list - like in this example:
> 
> ```
> seedwords='wall,input,chronic,gaze,tip,explain,fog,receive,kingdom,they,corn,universe,soup,roast,remind,country,make,harsh,dream,cross,all,magic,napkin,found'
> ```

There is a " " missing after the comma :/

It should be 
`seedwords='wall, input, chronic, gaze, tip, explain, fog, receive, kingdom, ...`

rootzoll, 2024-09-26:
@cstenglein comma separated should now contain also the space

TODO-TEST: With v1.11.3rc1 test fresh setup of cln node and check seed word listing.

closed 2024-11-07 by rootzoll