raspiblitz/raspiblitz #4708: [SetupFail] missing-passworda

rootzoll, 2024-08-19:
Setup of fatpack v1.11.2 fails with:
[image]

Idea would be to test if passworda is available (from storage) right after saving it and if it then later is not available to get/print more debug info.

cahueya, 2024-09-07:
I ran into the same bug with a 1.11.2-Installation. I set the Password A manually and reflashed the SD card, then it did recognize the already existing data and started the recovery process and I could set Password A new.

rootzoll, 2024-09-24:
Still not sure why this occurs sometimes. I added more debug info with #4750 that will be available from v1.11.3 by splittingh the error in missing-passworda-1 (the password does not get stored correctly by dialogPaswords.sh) and missing-passworda-2 (the password got lost or is not accessable).

Pushing to v1.12.0 and keeping the issue open for further reports.