raspiblitz/raspiblitz #4709: Tor exiting always after update to v1.11.2

BlueSnake830, 2024-08-26:
**Describe the bug**
I've updated to raspiblitz v1.11.2 from v1.10.0. After the update, TOR is not running again -> LND cannot activate the channels anymore

**To Reproduce**
Start raspiblitz v1.11.2

**Debug file**

sudo systemctl status tor
● tor.service - Anonymizing overlay network for TCP (multi-instance-master)
     Loaded: loaded (/lib/systemd/system/tor.service; enabled; preset: enabled)
     Active: active (exited) since Mon 2024-08-26 13:50:38 BST; 33min ago
   Main PID: 630 (code=exited, status=0/SUCCESS)
        CPU: 6ms

Aug 26 13:50:38 raspberrypi systemd[1]: Starting tor.service - Anonymizing overlay network for TCP (multi-instance-master)...
Aug 26 13:50:38 raspberrypi systemd[1]: Finished tor.service - Anonymizing overlay network for TCP (multi-instance-master).

rootzoll, 2024-09-26:
Can you turn off TOR and see if LND is finding peer?
SSHMENU > SETTINGS > "Run behind Tor" to not selected & OK

cstenglein, 2024-09-30:
Do you have LNDg installed by any chance?

A friend had a similar issue and the culprit was LNDg.

BlueSnake830, 2024-10-01:
I now found the root cause after starting from scratch. LNDg was not installed. 

My router received a new feature to block TOR traffic for specific VLANs. Unfortunately there is ab bug in the router firmware such that TOR is always blocked for all VLANs. 

So this is no bug in RaspiBlitz.