raspiblitz/raspiblitz #4711: Menu -> Connect -> mobile wallet - Zeus

fjone, 2024-08-27:
The genererated qr-code shows a core lightning-rest, which is now deprecated.
As of today the new Zeus wallet v0.9.0 have CLNrest backend support.

rootzoll, 2024-09-26:
@openoms what was the latest state of CLNrest - I mark this for next team meeting

openoms, 2024-09-27:
With the updated install script in https://github.com/raspiblitz/raspiblitz/pull/4748 now install the deps correctly and clnrest is running:
```
cl plugin list

      {
         "name": "/usr/local/libexec/c-lightning/plugins/clnrest/clnrest",
         "active": true,
         "dynamic": true
      },
```

The config needed so far only is only:
```
clnrest-port=<PORT_NUMBER>
```
in the clconf
https://docs.corelightning.org/docs/rest#configuration

rootzoll, 2024-10-01:
nmeeds the RTL update - to be tested

closed 2024-11-07 by rootzoll