raspiblitz/raspiblitz #4713: Wifi Connection Raspiblitz version 1.11.2

Haraldo90, 2024-08-29:
**Describe the bug**
Following the guideline from rootzoll I wanted to connect the Raspiblitz Raspberry Pi5 with the WLAN. 
I used a simple text note where I put in the first line my home Wifi and in the second line my password. 
Stored it with the name Wifi without .txt and flashed it in the first folder when oppening the PI SD card.
Now when I try to start the Rasperry PI I can not connect to my network and the following is written on the screen:
Failed to start _ a reverse proxy server 
Additionally there is no IP adress in the header of the Raspiblitz

**To Reproduce**
See above

**Debug file**
No file.

If possible, please add a debug report. To do that, drag a text file with the debug report in here.

mre-gi, 2024-08-30:
Hi @Haraldo90, 
I just experienced a similiar issue. The Raspiblitz wouldn't connect to my WIFI network. 
This is my first experience with the raspiblitz, so I'm a total newbie, so the following information is without any guarantee.

But it seems that the current way to set the wifi connection is not working, if your SSID consists of (white-)spaces. I managed to set the wifi connection manually by investigating the corresponding code. 

So if your SSID also contains spaces you can try the following:

1. Connect the Raspiblitz with the internet/LAN via RJ45 cable/port
2. Connect to the Raspiblitz via SSH
3. Abort the default SSH-Menu via Ctrl+C (exit to terminal)
4. Execute the following command. before replace the ${ssid} with your SSID-Name and ${password} with your WIFI-PW.
Command-Example: `/home/admin/config.scripts/internet.wifi.sh on ${ssid} ${password}`
Execution-Example: `/home/admin/config.scripts/internet.wifi.sh on 'My Personal WiFi' secretWifiPassword`

You can use this to check whether your raspiblitz can connect to your wifi at all. Please provide the console output if this does not work. However, this may not be the actual solution to your reverse proxy error. Nevertheless, the connection to the wifi is probably the prerequisite for the reverse proxy to work at all.

Source #1: https://github.com/raspiblitz/raspiblitz/issues/4384
Source #2: https://github.com/raspiblitz/raspiblitz/pull/4385/files

Haraldo90, 2024-08-30:
Hi @mre-gi,

thank you for the quick help. I'm a newbie too and glad for your support. Will try it out and let you know.
Many many thanks

Haraldo90, 2024-08-31:
Hi @mre-gi ,

your solution seems working. 
Many thanks for that, highly appreciate it.

Haraldo90, 2024-08-31:
Solution from @mre-gi works with version 1.11.2

closed 2024-08-31 by Haraldo90