raspiblitz/raspiblitz #4714: Failed message during start and loosing of connection with Full Node

Haraldo90, 2024-08-29:
**Describe the bug**
When I connect my Raspberry PI5 Raspiblitz via Ethernet cable and the system starts up there is the following error message.
(Failed)- Failed to start ..chnology (SMART) Daemon.
Seems for the first time no impact on the system cause normally the systems start regullary and searchs for Peers to Connect and start working. But after a while (30-45 min) the system is rebooting and not connecting again. Not sure if this belongs to the bug with the Smart Demon or what is the case for this.
Thx

**To Reproduce**
See above

**Debug file**
No file

rootzoll, 2024-09-26:
Sound like you getting "(Failed)- Failed to start Self-Monitoring, Analysis and Reporting Technology (SMART) Daemon" error which can happen if your connected drive is not S.M.A.R.T. capeable. Sounds like and old drive that might have hardware issues.

closed 2024-09-26 by rootzoll

FlyerOne86, 2024-11-21:
Same issue here since this week on the latest update. Samsung T7 SSD 1GB