raspiblitz/raspiblitz #4722: Update Core Lightning

rootzoll, 2024-09-02:
https://x.com/Core_LN/status/1830639492249567596

openoms, 2024-09-13:
The update from the menu works, but need to fix some errors:
```
remote: Total 148326 (delta 13846), reused 13622 (delta 13528), pack-reused 133731 (from 1)
Receiving objects: 100% (148326/148326), 104.05 MiB | 4.08 MiB/s, done.
Resolving deltas: 100% (98561/98561), done.
Updating files: 100% (27206/27206), done.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user
-action option if you know what you are doing and want to suppress this warning.
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'plugins/clnrest/requirements.txt'
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user
-action option if you know what you are doing and want to suppress this warning.
                                                                      
The Poetry configuration is invalid:
  - Additional properties are not allowed ('package-mode' was unexpected)
                                                                      
Submodule 'external/gheap' (https://github.com/valyala/gheap) registered for path 'external/gheap'
Submodule 'daemon/jsmn' (https://github.com/zserge/jsmn) registered for path 'external/jsmn'
```

rootzoll, 2024-09-24:
see PR #4748

closed 2024-10-04 by rootzoll