raspiblitz/raspiblitz #4727: [feat] Alby Hub

openoms, 2024-09-04:
https://blog.getalby.com/what-is-alby-hub/

https://github.com/getAlby/hub

bumi, 2024-09-05:
Let me know if I can help. Alby Hub is a single executable available for different environments.
As "node backend" it either works with a built in LDK based node or it can connect to a LND node (which could/should be the default for raspiblitz) If used with LND we can configure the address, cert, and macaroon with an env variable. 

no other config should be needed.

There is also a docker container if that helps.

Note: currently Alby Hub still requires an Alby account and has email notifications and similar enabled by default. In a future version this will be lifted.

MoritzKa, 2024-10-03:
Oh that would be cool! Then I can restart my Raspiblitz again.

boilroaming, 2024-10-04:
I really hope this will some soon.

rookey, 2024-10-12:
that would be awesome! I was somewhat confused why this doesn't work yet, as there's a menu item to connect to Getalby in Raspiblitz 1.11.x already. :)

itstomekk, 2024-10-12:
I see people asking for Alby Hub on Raspiblitz! +1

MindaugasN, 2024-10-17:
Looking forward for this 👍

rootzoll, 2024-11-18:
for a start I used `https://getalby.com/install/hub/pi-aarch64-install.sh)` and translated it into a the template install script:

https://github.com/raspiblitz/raspiblitz/blob/4727-albyhub/home.admin/config.scripts/bonus.albyhub.sh

Its in branch `4727-albyhub` to be tested and further developed:

- [ ] make also work when on x86 systems

rootzoll, 2024-11-20:
OK first PR is ready .. next orga TODOs:

- [ ] Test PR & squash-commit to `dev` branch
- [ ] Fork `v1.11` to `v1.11-albyhub` and cherry-pick PR
- [ ] Let users know how to become early-testers

And then finally for v1.12.0 release:

- [ ] Remove `(preview)` text
- [ ] Add to CHANGES

dragsugg, 2024-11-21:
Works excellent! Thank you! 😊

Kolomona, 2024-11-23:
Can this be fast tracked? The Alby-pocalypse is coming soon.

[image]

adevoss, 2024-11-25:
Switched from `dev` branch to `v1.11-albyhub` branch.
Gave me the spinning circle

Reinstall using `v1.11-albyhub` branch. All fine.

bumi, 2024-11-27:
@voidzero in Alby Hub the LND certificate does not get updated if LND changes that one. 
we have an update for this coming that makes sure always the updated data from the env variable is used.

voidzero, 2024-11-27:
What can I do in the mean time to fix this problem?

jpl-btc, 2024-11-27:
In case this folder or file is empty:  `.../app-data/albyhub/nwc.db` <- this one might be an empty file, check its size.

You might want to try this folder:
`.../app-data/alby-albyhub/data/albyhub/nwc.db`

At least that happens on an Umbrel, I'm not sure how folders are organized on your raspi

openoms, 2024-12-09:
Small issue:
got this during install
´´´
rm: cannot remove 'albyhub-server.tar.bz2': Permission denied
´´´
Will update with test findings.

rootzoll, 2024-12-12:
> rm: cannot remove 'albyhub-server.tar.bz2': Permission denied

see PR #4866 - that delete has a sudo now

steepdawn974, 2024-12-19:
Two (stylistic) comments:

1. @bumi I don't see any signature files in your releases. Raspiblitz should generally verify binaries after download before installing them... 
2. Imho there should not be a dependency on LND. AlbyHub could run with the embedded LDK node...

just my 50 sats

rootzoll, 2024-12-19:
@steepdawn974 I think it makes sense to offer AlbyHub on LND with RaspiBlitz - its a Full-Node and LND is one of its base Lightning implementations it provides (and people might already have channels with) - AlbyHub is an extension of the things you can then do with your LND.

To base AlbyHub on LDK would just make sense if it would be a genreal third Lightning Implementation on RaspiBlitz that you can choose on install, uses the local Bitcoin Core and can also provide services to other Apps in RaspiBlitz .. but I dont think thats the focus here and a too big of a change/task. 

If you really just want a lightweight AlbyHub with LDK you might not need all the Fullnode complexity of RaspiBlitz and possibly just install it plain on a RaspberryPi Zero: https://guides.getalby.com/user-guide/alby-account-and-browser-extension/hidden-archives/raspberry-pi-zero

bumi, 2024-12-19:
@steepdawn974 yes, important actually. and now that we are a bit more stable and have regular releases we need to add this. We'll make sure to have this with the next release

steepdawn974, 2024-12-19:
> @steepdawn974 I think it makes sense to offer AlbyHub on LND with RaspiBlitz - its a Full-Node and LND is one of its base Lightning implementations it provides (and people might already have channels with) - AlbyHub is an extension of the things you can then do with your LND.

Fair enough on the LDK node part.

However, since people are running with CLN instead of LND, it would be better if the installer detected which ones is running and/or asked the user which Lightning implementation they want as a backend to AlbyHub.

rootzoll, 2024-12-19:
> However, since people are running with CLN instead of LND, it would be better if the installer detected which one is running and/or asked the user which Lightning implementation they want as a backend to AlbyHub.

RaspiBlitz is able to run CLN & LND at the same time (only possible when managed by SSH menus) .. so if you run CLN and want to use AlbyHub you would can activate LND first and then have AlbyHub available.

I know its not perfect, but I really dont want another Lightningnode just running "on the side managed by an bonus app" on RaspiBlitz - this will introduce confusion on multiple levels. Like why is my Screen not showing my balance why were funds not backuped with my seed, etc ... @bumi best would be if AlbyHub also would support Core Lightning

bumi, 2024-12-19:
CLN support has been requested, but if nobody contributes it we don't have it on the roadmap, supporting more nodes means much more complexity to handle. We try to support LND as it's mostly adopted and optimize for simplicity with the embedded LDK.

steepdawn974, 2024-12-19:
@bumi I thought it is already supported, but mixed it up with the Alby Extension (where CLN is supported) .. sorry.

rootzoll, 2024-12-20:
OK we have a RaspiBlitz v1.11.4 Release Candidate with AlbyHUb (SSH & WebUI) for you to test .. any feedback over the weekend would be great: https://github.com/raspiblitz/raspiblitz/blob/v1.11/CHANGES.md#whats-new-in-version-1114-of-raspiblitz

tcmilleriii, 2024-12-24:
New here but hardware for my raspiblitz is en-route and I’d so love to get Alby Hub up and running with it!

rootzoll, 2024-12-24:
OK RaspiBlitz v1.11.4 with AlbyHub just got released: https://x.com/rootzoll/status/1871612490716688759

closed 2024-12-24 by rootzoll

MoritzKa, 2024-12-25:
Thanks for publishing Alby Hub. Super cool!!!

Just two questions regarding updates:
- How are new Alby Hub versions made available on the WebUI? Does the Alby team need to do anything in particular?
- Will users find a button on their WebUI if another Hub update is available? 

I think about writing a post on blog.getalby.com.

rootzoll, 2024-12-25:
> Thanks for publishing Alby Hub. Super cool!!!

:) Thx .. Alby Hub is a great functional & educational tool for lightning. Happy to have it on the RaspiBlitz

> How are new Alby Hub versions made available on the WebUI? Does the Alby team need to do anything in particular? Will users find a button on their WebUI if another Hub update is available?

So far there is no special update mechanism for AlbyHub. It will basically be our default way that when we do a new RaspiBlitz Release we will ship also updated versions of our bonus apps - like AlbyHub.

It would be great if you @MoritzKa from the AlbyHub team - when you release a new version of AlbyHub already open an issue on RaspiBlitz repo - letting us know about the new release and if there are any possible needed changes/migrations compared to the existing version (changes in config, setup process, database migrations, ...). Then we can add this to our next milestone planning.

Also signed released would be great for security in the future .. if you will have multiple people signing (e.g. in rotation) it would be good to know all of the possible valid signers so our install script will not fail if we hit an unknown signer.

Beside from that it would be great to have one named contact (github account name) that we can reffer in issues if a user comes with error reports to support us in figuring out if its on the RaspiBlitz or AlbyHube side and point into the right direction. Would then add that contact to the install script.