raspiblitz/raspiblitz #4740: LND won't start (v1.11)

Godrigos, 2024-09-15:
**Describe the bug**

A few days ago my raspiblitz (1.11) stop working properly. It now gets stuck in the "Node Syncing" screen with blockchain progress at 100% (several peers) and Lightning progress at prepeare sync, EMPTY, or a single time 100% but 0 peers !?...

The point is, I cant't reach SSH Menu to do something...

I don't know how to get logs, would appreciate some orientation to provide better debugging.
And the most important! Is there a way to at least recover funds?

Thanks for the attention.

openoms, 2024-09-23:
press CTRL+C ro get into the prompt at the terminal.
Type `menu` for the menu.
`blitzhelp` for help

Godrigos, 2024-09-24:
Thanks for reaching out, but once I was already thinking of changind my ssd and this happened. I just created another installation with version 1.11.2 and used SEED + SCB to recover. Unfortunatly I wasn't able to copy the blockchain (errors while trying to copy), so it is downloadind again.
Therefore, I won't be able to give you debugging information on this anymore, just hoping to have my funds back when the sync finishes.

closed 2024-09-24 by rootzoll

Godrigos, 2024-10-07:
Hi there, me again.
I will post here since it's a continuation of the same issue.
As said above I installed the newer version of raspiblitz in a new sd card. And entered SEED + SCB.
All seemed to be going well, the blockchain was downloading again...
Well, about a few days ago (blockchain was about 87%) the raspiblitz restarted by its own (asking for LN wallet password) and from that day on I only get:

`Blockcain progress: wainting for start
Lightining progress: prepare sync`

I already restarted and rebooted, this is all that I get.

[Here](https://termbin.com/r8bq) are the logs.

Any ideas?
Thanks for the help and patience.

Godrigos, 2024-11-29:
Hi. Sync processes is working fine on version 1.11.3. LND is now on recoveryscan  once I used SCB + SEED. Just a question, how long can it take to complete? It's been 13 days since the recovery began.