raspiblitz/raspiblitz #4745: Recommend 2TB SSD/NVMe

rootzoll, 2024-09-23:
even now before v1.12.0 in the doc
@openoms can you write here the info about the ssd board supporting 2TB ssd

openoms, 2024-09-23:
Did a basic recommendation here https://t.me/RaspiBlitzFAQ/802 I copy in:

# RPi4 disk update

To have safe power with 2TB disks I would use an X825 board (same as inside the metal cases):
https://www.amazon.com/Geekworm-Raspberry-X825-V2-0-compatible/dp/B091JSRQ9C

and power it through the barrel connector on the side with a 4A 5V power supply: https://www.amazon.de/gp/product/B088M1Q7GF

If already using an old X825 v1.5 board you will need to keep the USBC as well as the barrel connector, see this article: https://github.com/raspiblitz/raspiblitz/wiki/How-to-improve-the-RaspiBlitz-Metalcase-Powersupply

A good discussion starts here: https://t.me/raspiblitz/145920

rootzoll, 2024-09-26:
So this change would then go here: https://github.com/raspiblitz/raspiblitz-docs/blob/master/docs/setup/get-hardware.md

rootzoll, 2024-11-11:
OK we should start to pick/recommend some 2TB NVMe models for v1.12.0.

@openoms At the moment we hint at the "NVMe Crucial P3 Plus 2TB Gen4 CT2000P3PSSD8" which is (to my research) a QLC. Some people report high IO wait times with QLC and recommend to go at least TLC level.

| NAND Type | Performance      | Durability       | Price / GB      | Suitable for                                      |
|-----------|------------------|------------------|-----------------|---------------------------------------------------|
| SLC       | Very high        | Very high        | Very expensive  | Enterprise applications, critical systems         |
| MLC       | High             | High             | Expensive       | High-end SSDs, professional applications          |
| TLC       | Good             | Medium           | Medium          | Everyday SSDs, consumer market                    |
| QLC       | Medium to low    | Low              | Affordable      | Large storage, media, data storage                |

This could be TLC alternatives (need to be double checked):

- Kingston KC3000 PCIe 4.0 NVMe M.2 SSD
- Transcend SSD 2TB Transcend M.2 MTE115S
- Verbatim Vi3000 NVMe M.2 2 TB

List to be extened, checked and then sorted & (community) ranked.

NilByte, 2024-11-16:
Another resource of information on incompatibility:

https://wiki.geekworm.com/X1000

Some of the incompatibility seems to have been solved by firmware updates.

rootzoll, 2024-12-07:
Also cross-check with: https://pipci.jeffgeerling.com/#m2-and-nvme-adapters

rootzoll, 2025-01-04:
see: https://github.com/raspiblitz/raspiblitz-docs/issues/33#issuecomment-2571315040

rvullriede, 2025-02-23:
I'm currently looking for a stable HAT + SSD combination and want to share my findings so far:

Geekworm X1001 + Kingston KC3000: not stable (even with the default PCIe Gen 2) due to power problems. I'll test some other SSDs on the same board next.

https://community.volumio.com/t/pcie-nvme-compatibility/65722/2 also has a nice overview you might want to consider.

NilByte, 2025-02-24:
Extensive database here where various filters like low max. power can be set.

Example:

https://www.techpowerup.com/ssd-specs/search/?type=3&formfactor=2&interface=PCIe%203.0%20x4&powerMax=2&powerMaxRange=L&capacity=7

rootzoll, 2025-09-27:
OK because we recommend the Pimoroni NVMe base - I linked their compatibility list  for alternative NVMe models so far:
https://shop.pimoroni.com/products/nvme-base?variant=41219587178579

Marked for teamdiscuss for v1.12.1 if we wanna provide a own list or other list later on.