raspiblitz/raspiblitz #4760: putting this in bug report? BUT, since the hurricane passed though it has corrupted my main drive......

DrBlackross, 2024-09-27:
(didnt know where to put this since the issue format has changed since i bugged about the x86 stuff)

model name	: AMD A6-7400K Radeon R5, 6 Compute Cores 2C+4G
/dev/sda  main "blitz" system hard drive 28Gb
/dev/sdb  "node" hard drive with blockchain 2Tb

hey rootz.... if i reinstall the amd64 image on the main drive, the HD with the "node" info should be seen after being added before/after reboot?

or do i reinstall with the node HD in the system and it should see it and do a recovery? 

OR (last one) do i copy something off the node hd to my local system before reinstalling with the node HD and the main blitz hd... main drive is there as well as the system (i can log in) but i do not have full scp etc. all, well most, of the scripts are corrupt to do anything via ssh.

rootzoll, 2024-09-30:
So in this setup you would reinstall amd64 image on the system hard drive and it should on strat recognize the hard drive with blockchain and offer you the recover option. But this is still quite experimental ... RaspiBlitz v1.12.0 (the next release after 1.11.3) plans tp address/support better multiple HDD setups for laptops and VMs.

DrBlackross, 2024-09-30:
Cool, soooo i should wait till 1.12 release and let it limp along? lol

rootzoll, 2024-09-30:
If you can wait until Dec waiting for RaspiBlitz v1.12.0 could be an option.

DrBlackross, 2025-02-13:
safe to rebuild raspiblitz now? or still wait for 1.12?

DrBlackross, 2025-09-05:
cool looks like 1.12 is up and out (saw changes.md) now can i export my current raspiblitz setup and leave the original HD alone and just install 1.12 then import (via the thumbdrive) my previous raspibltz?

(yea mines been done a while lol)

rootzoll, 2025-09-27:
for the official release of v1.12.0 - check download section after the weekend: https://docs.raspiblitz.org/docs/setup/software-setup/download

closed 2025-09-27 by rootzoll

DrBlackross, 2025-09-29:
will that image work with x86? my setup was x86 after nespi case lol

rootzoll, 2025-09-29:
The image linked there is only for RaspberryPi .. for x86 you still need to install a basic debian12 and then run the build script manually.

DrBlackross, 2025-11-10:
ok, i've tried installing ubuntu server twice... (not a problem) following this page for amd64 to build the image
https://github.com/raspiblitz/raspiblitz/blob/dev/alternative.platforms/README.md#amd64-image
not a problem.. but i end up with 

nuggetnode:~$ sudo bash build_sdcard.sh -f false -b dev -d headless -t false -w off

'# updating locale ...
Error: 'ISO-8859-1' is not a supported language or locale
'*****************************************
'*     RASPIBLITZ BOOT IMAGE SETUP       *
'*****************************************
For details on optional parameters - call with '--help' or check source code.
interaction=true
fatpack=false
github_user=raspiblitz
branch=dev
display=headless
tweak_boot_drive=false
wifi_region=off
cpu=x86_64
architecture=amd64
\n# FAIL: Base image cannot be detected or is not supported.
PRETTY_NAME="Ubuntu 24.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.3 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
Linux nuggetnode 6.8.0-87-generic #88-Ubuntu SMP PREEMPT_DYNAMIC Sat Oct 11 09:28:41 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux

even after editing the build_sdcard.sh for the error 'ISO-8859-1' 
with

'# set locale to en_US.UTF-8 on system & activate for this script
if [ "$1" != "-EXPORT" ] && [ "$1" != "EXPORT" ]; then
  echo "# updating locale ..."
  sed -i "s/^# en_US.UTF-8 UTF-8.*/en_US.UTF-8 UTF-8/g" /etc/locale.gen
'#  sed -i "s/^# en_US ISO-8859-1.*/en_US ISO-8859-1/g" /etc/locale.gen
'#  locale-gen en_US.UTF-8 en_US ISO-8859-1 1>/dev/null
  locale-gen en_US.UTF-8 en_US 1>/dev/null
  update-locale LANG=en_US.UTF-8 1>/dev/null
  source /etc/default/locale
fi

i only have the Fail on the base image not detected

nuggetnode:~$ sudo bash build_sdcard.sh -f false -b dev -d headless -t false -w off
'# updating locale ...
'*****************************************
'*     RASPIBLITZ BOOT IMAGE SETUP       *
'*****************************************
For details on optional parameters - call with '--help' or check source code.
interaction=true
fatpack=false
github_user=raspiblitz
branch=dev
display=headless
tweak_boot_drive=false
wifi_region=off
cpu=x86_64
architecture=amd64
\n# FAIL: Base image cannot be detected or is not supported.
PRETTY_NAME="Ubuntu 24.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.3 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
Linux nuggetnode 6.8.0-87-generic #88-Ubuntu SMP PREEMPT_DYNAMIC Sat Oct 11 09:28:41 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux

hmmmm....

lol