raspiblitz/raspiblitz #4761: [Research] Packer or Docker/Ansible

rootzoll, 2024-09-27:
It looks like we fail to create fatpack with packer. So it this project could be another alternative for our CI and automated sd card image build: https://github.com/ptrsr/pi-ci

If we decide to stay witch packer the following issues are still on our todo list:
- [ ]  fatpack image packer build fails on local machine #4463 
- [ ]  CI: let fatpack image depend on lean image #4441 
- [ ]  CI: Cache node and other dependencies on image installs #4440 
- [ ]  [PACKER] Fatpack DOT net install/build fails #4707

openoms, 2024-09-28:
First step, forked the pi-ci repo and attempt to generate a Docker image based on the full RaspberryOS image (not lite): https://github.com/openoms/pi-ci/actions/runs/11087058497

woeisme, 2025-04-08:
@openoms I'd be interested in working on this with you too as I've been experimenting with raspiblitz and docker for a bit as well.