raspiblitz/raspiblitz #4765: LNBits installation displays empty url screen and no failure if PostgreSQL assertion was not met

cstenglein, 2024-09-30:
Tried to install btcpay & lnbits in one go.

the BTCPay installation failed because of the failed assertion below. It displayed a screen that the installation failed.

Afterwards LNBits was being installed and the same assertion failed (see below).

But then a screen appeared that LNBits is available under `http://` and not that the installation failed like with BTCPay.

```
FAIL - Was not able to start PostgreSQL service
○ postgresql@15-main.service - PostgreSQL Cluster 15-main
     Loaded: loaded (/lib/systemd/system/postgresql@.service; enabled; preset: enabled)
     Active: inactive (dead)
     Assert: start assertion failed at Mon 2024-09-30 13:00:56 BST; 30s ago
             AssertPathExists=/etc/postgresql/15/main/postgresql.conf was not met

Sep 30 12:58:16 raspberrypi systemd[1]: Starting postgresql@15-main.service - PostgreSQL Cluster 15-main...
Sep 30 12:58:18 raspberrypi systemd[1]: Started postgresql@15-main.service - PostgreSQL Cluster 15-main.
Sep 30 12:58:25 raspberrypi systemd[1]: Stopping postgresql@15-main.service - PostgreSQL Cluster 15-main...
Sep 30 12:58:25 raspberrypi systemd[1]: postgresql@15-main.service: Deactivated successfully.
Sep 30 12:58:25 raspberrypi systemd[1]: Stopped postgresql@15-main.service - PostgreSQL Cluster 15-main.
Sep 30 12:58:30 raspberrypi systemd[1]: postgresql@15-main.service: Starting requested but asserts failed.
Sep 30 12:58:30 raspberrypi systemd[1]: Assertion failed for postgresql@15-main.service - PostgreSQL Cluster >
Sep 30 13:00:56 raspberrypi systemd[1]: postgresql@15-main.service: Starting requested but asserts failed.
Sep 30 13:00:56 raspberrypi systemd[1]: Assertion failed for postgresql@15-main.service - PostgreSQL Cluster >
Failed to start lnbits.service: Unit lnbits.service not found.
```

rootzoll, 2024-09-30:
In PR #4768 I added a check if Postgres is already installed - preventing it to try to install again. This cleared the Assertion error on my lnbits install. @cstenglein please check if this also fixes your scenario

cstenglein, 2024-10-01:
> This cleared the Assertion error on my lnbits install

My issue was primary with the "success" screen on a unsuccessful installation:

[image]

Good that you found another issue there as well :+1:

cstenglein, 2024-10-01:
TODOs for @rootzoll:

- do the same fix for BTCPayServer
- add error screen if LNBits install fails

[image]

cstenglein, 2024-10-01:
Fix didn't work unfortunately 

```
psql: error: connection to server on socket "/var/run/postgresql/.s.PGSQL.5432" failed: No such file or directory
        Is the server running locally and accepting connections on that socket?
# postgresConfig failed -> SELECT datname FROM pg_database;
Failed to start lnbits.service: Unit lnbits.service not found.

rootzoll, 2024-10-03:
I tested in `4568-lnbits` branch on my x86 VM: in services SSH menu selected both BTCPay & LNBits to install .. and they both ran thru to install.

But it also was a fresh install .. is your scenario with old data in PostgresDB, maybe?

cstenglein, 2024-10-05:
One issue is at least, that the `off` command in the postgres script only uninstalls `postgresql` https://github.com/raspiblitz/raspiblitz/blob/dev/home.admin/config.scripts/bonus.postgresql.sh#L227C1-L227C32

The `on` command installs `postgresql-VERSION`, thus `postgresql-15`  https://github.com/raspiblitz/raspiblitz/blob/dev/home.admin/config.scripts/bonus.postgresql.sh#L35

cstenglein, 2024-10-05:
Tried to do the install with the new Scripts
Still didn't work but most likely the issue is on my side.

The screen for lnbits didn't display after the failed install tough, so that works now. 👍

closed 2024-10-06 by rootzoll