raspiblitz/raspiblitz #4775: connect Zeus (android)wallet SOCKS: Host unreachable

DuckG, 2024-10-02:
Hello,

I want to setup a new channel(connect to) a Zeus-android device. 
Zeus version (v9.0.0)
node version: v1.11.2.
Connect mobile wallet, zeus_android. Done

I can scan the qr-code,  After this it retrieves the node pubkey and host :9735
after using the 'open channel' button:
Error message "SOCKS: Host unreachable"

Is there a (easy)way to solve this error or to troubleshoot?
(Node is up and running and patched and already restarted.)

Or can anybody help me further?

Jomossey, 2024-10-02:
@DuckG it seems like you're running into a "SOCKS: Host unreachable" error when trying to open a channel on your Zeus-android device. One thing you could try is checking if there are any firewall settings that might be blocking the connection. Also, make sure the node pubkey and host details are correct.

DuckG, 2024-10-03:
i have found the problem or a bug and a workaround. 

First i used the Webportal, [connection details, Node id], show qr-code which by me didn't work, zeus-error-message 'SOCKS: Host unreachable' (LND-rest).
After this i tried to choose on the SSH-raspiblitz menu Connect, Mobile , Zeus_android. 
Then i scanned the code on my lcd-screen at the node. LND-(rest)/8080 Which is accepted/saved. [and solved]
So i don't understand why this doesn't work using the qr-code on the web-interface. Is this a Zues-scan-issue or do i misunderstand.  Maybe some-one can tell me?
For now by SSH and scanning the lcd display, the connection/channel is there.

How can i clean the lcd-display(the code is still there, also after closing ssh), except rebooting the node?