raspiblitz/raspiblitz #4777: [feat] CLN switch the pay plugin to renepay

openoms, 2024-10-04:
It can be done with a simple translator plugin taking over the pay RPC hook:

https://gist.github.com/openoms/8626623a124ff1164096f9b8d971195c

Tested so far with paying from 
- Zeus via CLNrest
- boltcard via LNbits

surfac3, 2024-10-04:
I tested this as well using both cmd line and Zeus (CLNrest) with CLN v24.08-60 on RaspiBlitz v1.11.2 

Payments ran smoothly with smaller amount(500k) and larger (1mil).

chrisguida, 2024-10-31:
There is also an "official" version of this plugin, for those interested: https://github.com/Lagrang3/default-renepay

openoms, 2024-11-01:
> There is also an "official" version of this plugin, for those interested: https://github.com/Lagrang3/default-renepay

Yes, it looks it does exactly the same. Forwards the `pay` RPC hook to `renepay`and matches the parameters. Bolt are quite simple, less than 100 lines in python. Feel free to test both.

openoms, 2024-11-01:
when sending with Zeus to a BTCPay lightningaddress getting the error with both plugins:
```
INFO    plugin-clnrest: RPC Error: {'code': -32602, 'message': 'bolt11 uses description_hash, but you did not provide description parameter'}
```

openoms, 2024-11-29:
The new CLN release will have this sorted:

>xpay is a new, experimental plugin for payments. It's rewritten from the ground up, on top of another plugin called askrene, which provides advanced routing advice for payments. The two play together: xpay tells askrene about the results of payment attempts, so it learns over time. The hard work was done by @Lagrang3 who wrote and refined the minimum-cost-flow solver which is the heart of askrene.
For reckless souls, xpay-handle-pay can be set to true (even at runtime, using setconfig) to have xpay take over the simple variants of pay.

Will close this issue after testing.

chrisguida, 2024-12-10:
@tobomobo

1) this issue is about `renepay`, not `xpay`
2) you need to set `xpay-handle-pay` as suggested here if you want `xpay` to take over for `pay`: https://github.com/ElementsProject/lightning/releases/tag/v24.11

rootzoll, 2025-05-17:
@openoms we are shipping RaspiBlitz v1.12.0 with Core Lightning v25.02.2 or above, so we can close this issue, correct?

geco91, 2025-05-26:
@openoms any todo for v1.12.0 here or push to later?

openoms, 2025-05-26:
xpay has been intruduced which does similar so closing this.