raspiblitz/raspiblitz #4779: CLN: missing `python3-grpcio` dependency

steepdawn974, 2024-10-04:
During install of the latest CLN 24.08.1 getting the (non fatal) error during the compilation phase:
`/usr/bin/python3: Error while finding module specification for 'grpc_tools.protoc' (ModuleNotFoundError: No module named 'grpc_tools')`

According to the docs  https://docs.corelightning.org/docs/installation the pip package `python3-grpcio-tool` is a dependency when building from source. 

Proposed Solution:
In [build_sdcard.sh](https://github.com/raspiblitz/raspiblitz/blob/dev/build_sdcard.sh#L378) add the apt package `python3-grpcio`  to `python_dependencies`.

This should solve it, but right now I don't have a test box to actually try it out and confirm, so I leave it as a bug report for now.

rootzoll, 2024-10-04:
@openoms can you take a look

openoms, 2024-10-04:
Please try to reinstall CLN with the updated cl.install.sh in the dev branch. Should cover all deps with poetry.

steepdawn974, 2024-10-04:
> Please try to reinstall CLN with the updated cl.install.sh in the dev branch. Should cover all deps with poetry.

Hmm, that's what I had done...
 `cl.update.sh reckless`  on dev (93777eda4d685b67ff4a812f1f3a3b87ca461c89) which does call cl.install.sh

But yes, if apt python3-grpcio needs to be added, then not in `build_sdcard.sh` as I suggested above but rather in `cl.install.sh` > installDependencies()

steepdawn974, 2024-10-04:
I went through the cl.* scripts again, and the CLN pyproject.toml , and yes, it should call poetry and poetry should install `pyhton3-grpcio-tools` as @openoms suggested.. 

Now that I have manually installed the apt package `python3-grpcio`, I can't reproduce the old scenario anymore :(

When I find some time, I'll bring up a new blitz VM and test/confirm from scratch.

rootzoll, 2024-10-04:
> But yes, if apt python3-grpcio needs to be added, then not in build_sdcard.sh as I suggested above but rather in cl.install.sh > installDependencies()

So should we add `python3-grpcio` here for v1.11.3 ? https://github.com/raspiblitz/raspiblitz/blob/93777eda4d685b67ff4a812f1f3a3b87ca461c89/home.admin/config.scripts/cl.install.sh#L35-L36

openoms, 2024-10-04:
@rootzoll no we shouldn't, the updated install script in dev covers it with poetry.

openoms, 2024-10-08:
recovery on v1.11.3rc1 looking good:
```
₿ cat raspiblitz.provision-update.log | grep grpc
  - Installing grpcio (1.64.1)
  - Installing pyln-grpc-proto (0.1.2 /home/bitcoin/lightning/contrib/pyln-grpc-proto)
  - Installing grpcio-tools (1.62.2)
msggen cln-grpc/proto/node.proto
python3 -m grpc_tools.protoc -I cln-grpc/proto cln-grpc/proto/node.proto --python_out=contrib/pyln-grpc-proto/pyln/grpc/ --grpc_python_out=contrib/pyln-grpc-proto/pyln/grpc/ --experimental_allow_proto3_optional
msggen cln-grpc/proto/node.proto
+grpc-msg-buffer-size=
+grpc-port=
```

rootzoll, 2024-10-08:
Perfect. Closing this for RaspiBlitz v1.11.3 release.

closed 2024-10-08 by rootzoll