raspiblitz/raspiblitz #4781: Stuck WebUi running Raspiblitz on Proxmox with Debian 11

MarsRzR, 2024-10-04:
**Describe the bug**
Opening the WebUI works but only shows the white/yellow Raspiblitz logo with the loading icon below
Even after full sync of LND and BTC the WebUI does not get available
Bugreport shows issue with BlitzAPI
Reinstalling API and WebUI does not change the described behavior.

**Debug file**
[BugReport.txt](https://github.com/user-attachments/files/17260624/BugReport.txt)

rootzoll, 2024-10-08:
You are using blitzversion: 1.11.2 .. we just released v1.11.3rc2 in (latest in the `dev` branch) which contains a lot of fixes to esspecially for Debian12 based VMs on proxmox. Would recommend to give that a try.

MarsRzR, 2024-10-17:
Hi @rootzoll,
thanks for the update.
I will give it a try and provide feedback 👍

MarsRzR, 2024-11-17:
Hi @rootzoll,
I finally found time to update to 1.11.3
Unfortunately the WebUI is still stuck showing the same issue as before.
As shown in attached bug report, the Blitzapi.service exited with following log entry:

Nov 17 23:22:57 raspiblitz python[141210]: /home/blitzapi/blitz_api/venv/bin/python: No module named uvicorn

Debug file
[BugReport.txt](https://github.com/user-attachments/files/17792645/BugReport.txt)

MarsRzR, 2024-11-17:
Hi @rootzoll,

after one negative try with Debian 11 I got it running on Debian 12.
thanks for the support!

closed 2024-11-17 by MarsRzR