raspiblitz/raspiblitz #4784: CPU +100% and disk 99% full: need to migrate?

gittbuh, 2024-10-05:
Hi all,

My raspiblitz on a pi4 4G with the recommended hardware online since 2020 is running at +100% CPU and using 99% of the 1T ssd disk space. What shall I do? Is this normal? Do I need to migrate?

rootzoll, 2024-10-15:
If you disk is full you need to change to new hardware - 2 TB SSD recommended, also RaspberryPi 5. BUT with a disc 99% full this seems to be quite a difficult task, because first you would need to free space so that node can create migration file and so on and so on .... I would recommend the follwoing:

Wait until RaspiBlitz v1.12.0 (around end of 2024) and then get the latest recommended hardware. RaspiBlitz v1.12.0 will introduce improved migration features so that maybe you can just connect your old SSD to the new RaspberryPi5 and let it run, etc ... but that is still in the works.

gittbuh, 2024-10-21:
Thanks a lot for the reply and recommendation. I spend most of the year very far away from the hardware my node is running on and for this reason I would like it to be more resilient. I am therefore planning to move to a VM in Proxmox. What do you recommend I do in this case? Can I just close channels, move funds on-chain and send them all out (empty my current Raspiblitz node) and start a new one in a Proxmox VM? Any major problem with this route?

rootzoll, 2024-11-12:
Starting fresh is always a good start .. I hope we can provide some good migration options to move from RaspberryPi to Proxmox with RaspiBlitz with future RaspiBlitz v1.12.0.

rootzoll, 2025-05-18:
RaspiBlitz v1.12 that is getting into final preperation has now a "FROMHDD" migration option during setup/recovery offering to copy from a smaller drive to a bigger drive that are both connected to a RaspberryPi. 

Migration options to Proxmox will be further explored in upcomming 1.12.x verisons of RaspiBlitz.