raspiblitz/raspiblitz #4792: v1.11.3rc1 LND unlock issue

cstenglein, 2024-10-07:
Set up the 1.11.3rc1 successfully

Now it asks me for password C to unlock the LND wallet.

I enter it, it says "OK wallet unlocked" and continues into the sync screen.

After the next background loop, it displays the unlock screen again.

cstenglein, 2024-10-07:
LND remains in `Not Started | Not Ready Yet | No Data` state when going in through the menu.

Debug log:
[debug.txt](https://github.com/user-attachments/files/17281162/debug.txt)

rootzoll, 2024-10-07:
This seems to be a problem that your LND is not able to start thru after unlock, it restarts and needs wallet again to be unlocked:

```
2024-10-07 16:27:10.149 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2024-10-07 16:27:12.156 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2024-10-07 16:27:13.808 [INF] LNWL: Opened wallet
2024-10-07 16:27:14.164 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: the RPC server is in the process of starting up, but not yet ready to accept calls
2024-10-07 16:27:14.201 [ERR] LTND: unable to create partial chain control: unable to determine if bitcoind is pruned: unable to detect bitcoind version: json: cannot unmarshal array into Go struct field GetNetworkInfoResult.warnings of type string
2024-10-07 16:27:14.201 [ERR] LTND: Shutting down because error in main method: error creating wallet config: unable to create partial chain control: unable to determine if bitcoind is pruned: unable to detect bitcoind version: json: cannot unmarshal array into Go struct field GetNetworkInfoResult.warnings of type string
2024-10-07 16:27:14.236 [INF] LTND: Shutdown complete
```

LND is even unable to detect the Bitcoin Core version ... and its not alone with that - also the API is failing to parse info from Bitcoin Core:

```
Oct 07 16:42:38 raspberrypi python[1221]:   File "/home/blitzapi/blitz_api/app/bitcoind/service.py", line 102, in get_network_info
Oct 07 16:42:38 raspberrypi python[1221]:     if result["error"] is not None:
Oct 07 16:42:38 raspberrypi python[1221]:        └ {'jsonrpc': '2.0', 'result': {'version': 280000, 'subversion': '/Satoshi:28.0.0/', 'protocolversion': 70016, 'localservices':...
Oct 07 16:42:38 raspberrypi python[1221]: KeyError: 'error'
Oct 07 16:42:38 raspberrypi python[1221]:
Oct 07 16:42:38 raspberrypi python[1221]: 2024-10-07 16:42:38 | ❌ | service.py:136 | Blockchain Info or Network Info not available: None None
Oct 07 16:42:38 raspberrypi python[1221]: 2024-10-07 16:42:38 | ❌ | service.py:210 | Unable to get blockchain or network data. See the log files for more information
```

Do you already updated Bitcoin Core to `'version': 280000,`?

redmac68, 2024-10-07:
Just in case it is of any use - I updated my bitcoin core on Raspiblitz
v1.11.1 to 28.0 and I got the same or very similar error and LND went into
a shutdown/restart loop. Resolved by reverting to Bitcoin core 27.1.

cstenglein, 2024-10-07:
> This seems to be a problem that your LND is not able to start thru after unlock, it restarts and needs wallet again to be unlocked:
>
> ```
> 2024-10-07 16:27:10.149 [ERR] RPCS: [...]

Yeah looks correct that I somehow use bitcoind version 28.0.

Tried to use `./config.scripts/bitcoin.install.sh off` to uninstall and then `./config.scripts/bitcoin.install.sh on` to re-install bitcoind.

But it looks to install 28.0 by default:

```
admin@<myip>:~ ₿ ./config.scripts/bitcoin.install.sh on
# Running: bitcoin.install.sh on
# Installing Bitcoin Core instance on mainnet
# /home/bitcoin/.bitcoin/bitcoin.conf is present
rpcallowip=127.0.0.1
main.rpcbind=127.0.0.1:8332
main.rpcport=8332
main.zmqpubrawblock=tcp://127.0.0.1:28332

[Unit]
Description=Bitcoin daemon on mainnet

Wants=redis.service
After=redis.service

[Service]
Environment='MALLOC_ARENA_MAX=1'
ExecStartPre=-/home/admin/config.scripts/bitcoin.check.sh prestart mainnet
ExecStart=/usr/local/bin/bitcoind  \
                                  -daemonwait \
                                  -conf=/mnt/hdd/bitcoin/bitcoin.conf \
                                  -datadir=/mnt/hdd/bitcoin
PermissionsStartOnly=true

# Process management
####################
Type=forking
Restart=on-failure
TimeoutStartSec=infinity
TimeoutStopSec=600

# Directory creation and permissions
####################################
# Run as bitcoin:bitcoin
User=bitcoin
Group=bitcoin

StandardOutput=null
StandardError=journal

# Hardening measures
####################
# Provide a private /tmp and /var/tmp.
PrivateTmp=true
# Mount /usr, /boot/ and /etc read-only for the process.
ProtectSystem=full
# Deny access to /home, /root and /run/user
ProtectHome=true
# Disallow the process and all of its children to gain
# new privileges through execve().
NoNewPrivileges=true
# Use a new /dev namespace only populated with API pseudo devices
# such as /dev/null, /dev/zero and /dev/random.
PrivateDevices=true
# Deny the creation of writable and executable memory mappings.
MemoryDenyWriteExecute=true

[Install]
WantedBy=multi-user.target

Created symlink /etc/systemd/system/multi-user.target.wants/bitcoind.service → /etc/systemd/system/bitcoind.service.
# OK - the bitcoin daemon on mainnet service is now enabled
# Add aliases bitcoin-cli, bitcoinlog
alias bitcoin-cli="sudo -u bitcoin /usr/local/bin/bitcoin-cli -rpcport=8332"
alias bitcoinlog="sudo -u bitcoin tail -n 30 -f /mnt/hdd/bitcoin/debug.log"
alias bitcoinconf="sudo nano /mnt/hdd/bitcoin/bitcoin.conf"
# OK - the bitcoind.service is enabled, system is ready so starting service
# Installed Bitcoin Core version v28.0.0

# Monitor the bitcoind with:
# sudo tail -f /mnt/hdd/bitcoin/debug.log
```

See the line at the end `# Installed Bitcoin Core version v28.0.0`

cstenglein, 2024-10-07:
Installing removing bitcoin 28.0 and installing bitcoin 26.0 with `./config.scripts/bitcoin.install.sh install` fixed the issue for me, thank you @rootzoll  & @redmac68  :+1: 

@rootzoll please check if bitcoin 28.0 was maybe already included in the rc1. I didn't set anything up myself on there.

rootzoll, 2024-10-08:
> please check if bitcoin 28.0 was maybe already included in the rc1. I didn't set anything up myself on there.

Its not included - but if once Bitcoin Core was updated using `reckless` on next recover/update it will take the latest Bitcoin Core which is now v28 that breaks with LND & API - this also effects people now updating to RaspiBlitz v1.11.2 with a reckless setting.

As a quick fix I will set downgrade `reckless` to `tested` max installing Bitcoin Core v27.1

rootzoll, 2024-10-08:
I tried the `-deprecatedrpc=warnings` flag, but it wuns into the same problems.

As to see - the flag is activated

```
*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: active (running) since Tue 2024-10-08 12:31:39 BST; 19min ago
   Main PID: 7663 (bitcoind)
      Tasks: 19 (limit: 4454)
        CPU: 5min 33.194s
     CGroup: /system.slice/bitcoind.service
             └─7663 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -deprecatedrpc=warnings
```

But still error with LND and BitzAPI - see:
```
*** LAST BLITZAPI LOGS ***
sudo journalctl -u blitzapi -b --no-pager -n20
Oct 08 12:50:15 raspberrypi python[998]:     self._context.run(self._callback, *self._args)
Oct 08 12:50:15 raspberrypi python[998]:     │    │            │    │           │    └ <member '_args' of 'Handle' objects>
Oct 08 12:50:15 raspberrypi python[998]:     │    │            │    │           └ <Handle <TaskStepMethWrapper object at 0x7fa53fd7b0>()>
Oct 08 12:50:15 raspberrypi python[998]:     │    │            │    └ <member '_callback' of 'Handle' objects>
Oct 08 12:50:15 raspberrypi python[998]:     │    │            └ <Handle <TaskStepMethWrapper object at 0x7fa53fd7b0>()>
Oct 08 12:50:15 raspberrypi python[998]:     │    └ <member '_context' of 'Handle' objects>
Oct 08 12:50:15 raspberrypi python[998]:     └ <Handle <TaskStepMethWrapper object at 0x7fa53fd7b0>()>
Oct 08 12:50:15 raspberrypi python[998]:   File "/home/blitzapi/blitz_api/app/bitcoind/service.py", line 204, in _handle_gather_bitcoin_status
Oct 08 12:50:15 raspberrypi python[998]:     info = await get_btc_info()
Oct 08 12:50:15 raspberrypi python[998]:                  └ <function get_btc_info at 0x7fa618d120>
Oct 08 12:50:15 raspberrypi python[998]: > File "/home/blitzapi/blitz_api/app/bitcoind/service.py", line 133, in get_btc_info
Oct 08 12:50:15 raspberrypi python[998]:     ninfo = await get_network_info()
Oct 08 12:50:15 raspberrypi python[998]:                   └ <function get_network_info at 0x7fa618c9a0>
Oct 08 12:50:15 raspberrypi python[998]:   File "/home/blitzapi/blitz_api/app/bitcoind/service.py", line 102, in get_network_info
Oct 08 12:50:15 raspberrypi python[998]:     if result["error"] is not None:
Oct 08 12:50:15 raspberrypi python[998]:        └ {'jsonrpc': '2.0', 'result': {'version': 280000, 'subversion': '/Satoshi:28.0.0/', 'protocolversion': 70016, 'localservices':...
Oct 08 12:50:15 raspberrypi python[998]: KeyError: 'error'
Oct 08 12:50:15 raspberrypi python[998]:
Oct 08 12:50:15 raspberrypi python[998]: 2024-10-08 12:50:15 | ❌ | service.py:136 | Blockchain Info or Network Info not available: None None
Oct 08 12:50:15 raspberrypi python[998]: 2024-10-08 12:50:15 | ❌ | service.py:210 | Unable to get blockchain or network data
```

Will fall back to downgrade `reckless` to `tested` on Bitcoin Core.

rootzoll, 2024-10-08:
OK so if people run into the "LND unlock" issue because of an reckless update to Bitcoin Core v28 - a fix will be to install RaspiBlitz v1.11.3rc2 or final that will downgrade reckless to tested Bitcoin Core v27.1 and not break things.

openoms, 2024-10-09:
I have added:
```
deprecatedrpc=warnings
```
to the bitcoin.conf and restarted bitcoind.
Works after that.

rootzoll, 2024-10-15:
@openoms LND v0.18.4 id a major version jump from LND 0.17.5 that we are using now and would need to compiled adaptors .. to not further delay the Release v1.11.3 I would recommend we go with the limited reckless update and dont overwrite values in the `bitcoin.conf` .. with RaspiBlitz v1.12.0 we can then upgrade Bitcoin Core and all the others and we should be good.

ferdinandhodl, 2025-05-17:
btw. [`lncli lock` command](https://github.com/lightningnetwork/lnd/issues/1162) would also be a welcome enhancement

rootzoll, 2025-05-17:
Raspiblitz 1.12.0 will ship with LND v1.18.,5 or above - so closing this issue.

closed 2025-05-17 by rootzoll