raspiblitz/raspiblitz #4798: gRPC Settings

bvolution, 2024-10-14:
**Describe the bug**
I am observing some strange behavior after I ran `sudo apt-get update && sudo apt-get upgrade -y`
I am currently not able to login into lndg, which fails with the following error message:
```sh
An error has occured!
LNDg cannot reach the gRPC for LND. Make sure LND is running and/or your gRPC settings are correct.
```

**Question**
How / where can I check the gRPC settings? I noticed that during the sys update / upgrade I was asked about `/etc/pam.d`, if this should be overwritten, I chose no, as I was unsure if this is going to hurt my system. How could I fix gRPC if I broke it somehow?

rootzoll, 2024-10-15:
Best would be to have a debug log to check the state of your system after the upgrade: https://docs.raspiblitz.org/docs/faq#debug

bvolution, 2024-10-15:
https://termbin.com/xidj

bvolution, 2024-10-15:
I am sorry, to have bothered you.... restart did fix it.

closed 2024-10-15 by bvolution