raspiblitz/raspiblitz #4799: PublicPool - Timezone

rootzoll, 2024-10-15:
> Public Pool seems to require that the server time zone is set to UTC to display graphs with your local time within the graphs refresh rate.

As mentioned here:
https://github.com/raspiblitz/raspiblitz/pull/4785#issuecomment-2405729226

```
admin@192.168.178.143:~ ₿ timedatectl
Local time: Tue 2024-10-15 12:45:52 BST
Universal time: Tue 2024-10-15 11:45:52 UTC
RTC time: Tue 2024-10-15 11:45:52
Time zone: Europe/London (BST, +0100)
System clock synchronized: yes
NTP service: active
RTC in local TZ: no
```

@NilByte My testnode runs not on UTC and I get graph data. So its not functional critical, correct? 

I mean we could set the whole RaspiBlitz to UTC - but that would not be the correct setting for the node. So from the first look this seems to be more of an issue for the PublicPool project to address.

NilByte, 2024-10-15:
@rootzoll correct that it doesn't seem to be functional critical.

During my initial test, I had time zone set to UTC +2 and it just resulted in the fact that the time for hash graph was 2 hours off when you hover the data points logged every 10 minutes. The initial test i did on UTC +2 didn't affect shares being accepted as valid in any way.

Setting the time zone to UTC made the 10 minutes data points for workers match my local time but on the other hand there seems to be some strange 2 hour backdating on the High Scores in main page if the time zone set to UTC, so it looks inconsistent to me how public pool handle time somehow.

NilByte, 2024-11-03:
https://github.com/benjamin-wilson/public-pool-ui/issues/20#issue-2630906515

NilByte, 2025-02-28:
https://github.com/benjamin-wilson/public-pool/pull/96

NilByte, 2025-07-08:
Both Raspiblitz and Public Pool support local time zone set in Raspiblitz.

It's just in Public Pool High Scores the displayed time will be backdated if you set a local time zone.

In:
src/app/components/splash/splash.component.html

 Find:
`<td>{{ highScore.updatedAt | dateAgo }} ago</td>`

Replace with:
`<td>{{ highScore.updatedAt + 'Z' | dateAgo }} ago</td>`

Not sure why Public Pool don't just fix it.