raspiblitz/raspiblitz #4805: Node Keeps Restarting, Won't Complete Blockchain Sync

jollypatron, 2024-10-17:
Hi Everyone,

I keep running into the same problem with my node. It starts its initial sync of the blockchain, and then after about 5-7% through, it restarts over and over and doesn't make any more progress.

[image]

I've looked through some of the similar issues posted on the forum, but I haven't been able to find anything that works. I even tried starting my node again from scratch and I still get this message. Here is the debug log. Really appreciate any help you can provide!

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.11.1
chainnetwork: bitcoin / test
 03:35:27 up  7:41,  4 users,  load average: 1.44, 1.55, 1.57

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT                  LOAD   ACTIVE SUB    DESCRIPTION
● bitcoind.service      loaded failed failed Bitcoin daemon on mainnet
● smartmontools.service loaded failed failed Self Monitoring and Reporting Technology (SMART) Daemon

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
2 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
× bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2024-10-17 00:52:45 BST; 2h 42min ago
   Duration: 4h 58min 14.771s
    Process: 1932718 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 1932732 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=1/FAILURE)
        CPU: 141ms

Oct 17 00:52:45 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 17 00:52:45 raspberrypi systemd[1]: Failed to start bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
Oct 17 00:52:44 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 5.
Oct 17 00:52:44 raspberrypi systemd[1]: Stopped bitcoind.service - Bitcoin daemon on mainnet.
Oct 17 00:52:44 raspberrypi systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Oct 17 00:52:44 raspberrypi bitcoin.check.sh[1932719]: /bin/chgrp: changing group of '/mnt/hdd/bitcoin': Read-only file system
Oct 17 00:52:44 raspberrypi bitcoin.check.sh[1932720]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Oct 17 00:52:44 raspberrypi bitcoin.check.sh[1932721]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Oct 17 00:52:44 raspberrypi bitcoin.check.sh[1932722]: grep: /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Oct 17 00:52:44 raspberrypi bitcoin.check.sh[1932723]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Oct 17 00:52:44 raspberrypi bitcoin.check.sh[1932724]: touch: cannot touch '/mnt/hdd/bitcoin/debug.log': Read-only file system
Oct 17 00:52:44 raspberrypi bitcoin.check.sh[1932725]: chown: changing ownership of '/mnt/hdd/bitcoin/debug.log': Read-only file system
Oct 17 00:52:44 raspberrypi bitcoin.check.sh[1932726]: chmod: changing permissions of '/mnt/hdd/bitcoin/debug.log': Read-only file system
Oct 17 00:52:44 raspberrypi bitcoind[1932732]: Error: Error reading configuration file: specified config file "/mnt/hdd/bitcoin/bitcoin.conf" could not be opened.
Oct 17 00:52:44 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Oct 17 00:52:44 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 17 00:52:44 raspberrypi systemd[1]: Failed to start bitcoind.service - Bitcoin daemon on mainnet.
Oct 17 00:52:45 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 6.
Oct 17 00:52:45 raspberrypi systemd[1]: Stopped bitcoind.service - Bitcoin daemon on mainnet.
Oct 17 00:52:45 raspberrypi systemd[1]: bitcoind.service: Start request repeated too quickly.
Oct 17 00:52:45 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 17 00:52:45 raspberrypi systemd[1]: Failed to start bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
[... 217 lines cut ...]
system_up="27698"
system_board="rp5"
system_ram_gb="8"
system_init_time="1729104851"
ln_default_sync_initial_done=""
btc_default_mempool_transactions="0"
btc_default_sync_initialblockdownload="1"
btc_default_sync_progress="0.07498573968822972"
btc_default_blocks_behind="494319"
btc_default_sync_percentage="7.49"
btc_default_peers="10"
btc_default_running="0"
btc_default_sync_initial_done="0"
btc_default_blocks_data_kb="48582356"
btc_default_online="0"
btc_default_blocks_headers="865968"
btc_default_port="8333"
btc_default_ready="0"
btc_default_synced="0"
btc_default_blocks_verified="371649"
btc_default_version="v26.0.0"
btc_default_error_short=""
btc_default_activated="1"
btc_default_error_full=""
*** POSSIBLE ERROR REPORTS ***

mrpaper87, 2024-10-25:
Same problem:
[image]

jollypatron, 2024-10-25:
I did some more digging, and I found that it might be an issue with the amount of power that is being delivered to my SSD via the raspberry PI. If not enough power is getting to SSD, it can impact its ability to read/write. This can especially happen if you're not using the recommended SSD from the documentation (which I am not).

I started the entire process over again (re-flashed micro SD). I also went out and got a USB power hub so that the SSD has its own power source and it's not solely reliant on the raspberry pi. This is the one I got on Amazon: https://www.amazon.com/dp/B00TPMEOYM?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

Good news is that it seems to be working now! Sync is still in progress and not complete yet, but it's made it a lot further than it normally does. Will update this thread if it completes properly.

I've seen a lot of similar posts on here of people with the same issue, and it's not always a power issue. But I think this is what it was for me...hopefully this helps!

mrpaper87, 2024-10-30:
Thanks for your answer. I changed the disk adapter and now it seems to work.

jollypatron, 2024-10-31:
Great! It was the power to my ssd that was my issue. Blockchain now fully synced without any error after using USB hub to power ssd.

closed 2024-10-31 by jollypatron