raspiblitz/raspiblitz #4817: BTCPayInstall fails on Update/Recovery

rootzoll, 2024-10-25:
I had a RaspiBlitz v1.11.3 (d24feead) BTCPay Server installed and run recovery with RaspiBlitz v1.11.3 minimal sd card image.

After recovery when I first time selected "BTCPayServer" in the SSH menu I got "BTCPay Server needs to be re-installed.\nPress OK to start process." - see this line here: https://github.com/raspiblitz/raspiblitz/blob/c5def56588acce308bb643ef4eb1772fcb3135af/home.admin/config.scripts/bonus.btcpayserver.sh#L231-L232

And then it ends up in a loop saying "# Waiting for nbxplorer to start - CTRL+C to abort"

[image]

@openoms can you take a look?

openoms, 2024-10-25:
couldn't reproduce this so far, nbxplorer and btcpay are running ok after a fresh install and update also.

When this happens can you please check:
```
sudo journalctl -fu nbxplorer -n50
```
?

cdemkos, 2024-10-25:
[image]
This is the result of "sudo journalctl -fu nbxplorer -n50"

rootzoll, 2024-10-25:
@openoms when testing #4819 it slaso fixed the BTCPay install ... so maybe this was a side effect.

openoms, 2024-10-26:
> ![Bildschirmfoto vom 2024-10-25 22-42-38](https://private-user-images.githubusercontent.com/152734664/380320870-6609998a-cfec-41c0-b885-cc035bd65015.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk5MzYzMjIsIm5iZiI6MTcyOTkzNjAyMiwicGF0aCI6Ii8xNTI3MzQ2NjQvMzgwMzIwODcwLTY2MDk5OThhLWNmZWMtNDFjMC1iODg1LWNjMDM1YmQ2NTAxNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyNlQwOTQ3MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZDdlY2RhOGYxODQ3MGMxMWJmNzcwZGZhY2NmMzc2YzZjMzcyZWNjODIyY2Q5ODdjNTY1M2E5MmM5MGQyNWZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.RJFZjGEb7pppJnSR5Kn1Q18yYCmGTU7AsBUWYoPczE4) This is the result of "sudo journalctl -fu nbxplorer -n50"

This looks like there was a problem downloading or installing the dotnet framework, could be related to a temporary connection issue a well.
We should ensure that the install script fails more clearly.

closed 2024-10-28 by rootzoll

rootzoll, 2024-10-29:
OK .. this error seems a bit tricky because it sometimes happens and sometimes not .. its with the cloning on the NBexplorer github repo and sometimes the repo just failes to clone and it eneded up in inbetween state.

It now at least rollsback the BTCPay Server install if this happens - see latest merges in dev & v1.11 branches. Or test with v1.11.3rc4 images - see https://github.com/raspiblitz/raspiblitz/blob/dev/CHANGES.md#whats-new-in-version-1113-of-raspiblitz

commit 7c91ee99 (2024-10-28): Fix #4817 BTCPay Server install (#4824)
commit 5bb31130 (2024-10-28): Fix #4817 BTCPay Server install (#4824)
commit a64e31f7 (2024-10-28): Fix #4817 BTCPay Server install (2) (#4825)

* #4817 add fallback
commit 083c3331 (2024-10-28): Fix #4817 BTCPay Server install (2) (#4825)

* #4817 add fallback