raspiblitz/raspiblitz #4818: LND signer PGP key changed

openoms, 2024-10-25:
on recovery using the lean image was missing the lnd binary.

Running 
`./config.scripts/lnd.install.sh install`
results in:

```
File ‘manifest-roasbeef-v0.17.5-beta.sig’ not modified on server. Omitting download.

WARNING: timestamping does nothing in combination with -O. See the manual
for details.

--2024-10-25 20:36:32--  https://keybase.io/roasbeef/pgp_keys.asc
Resolving keybase.io (keybase.io)... 3.87.151.43, 3.230.6.109
Connecting to keybase.io (keybase.io)|3.87.151.43|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1306 (1.3K) [text/plain]
Saving to: ‘pgp_keys.asc’

pgp_keys.asc                                              100%[====================================================================================================================================>]   1.28K  --.-KB/s    in 0s      

2024-10-25 20:36:33 (23.4 MB/s) - ‘pgp_keys.asc’ saved [1306/1306]

pub   ed25519 2024-10-21 [SC]
      A5B61896952D9FDA83BC054CDC42612E89237182
uid                      Olaoluwa Osuntokun <laolu32@gmail.com>
sub   cv25519 2024-10-21 [E] [expires: 2025-10-21]
sub   ed25519 2024-10-21 [A] [expires: 2025-10-21]
sub   ed25519 2024-10-21 [S] [expires: 2025-10-21]

# BUILD WARNING --> LND PGP author not as expected
Should contain PGP: E4D85299674B2D31FAA1892E372CBD7633C61696
PRESS ENTER to TAKE THE RISK if you think all is OK
```

openoms, 2024-10-25:
Looks like Laolu has changed his key which is now different from what he used to sign the v0.17.5-beta release.

rootzoll, 2024-10-25:
Good that happened before final v1.11.3 release

PR #4819 (2024-10-25): fix ln signing key, use guggero for old release, update roasbeef key
FIxes: #4818
closed 2024-10-25 by rootzoll