raspiblitz/raspiblitz #4822: Install failed-> "fatal: fetch-pack:  invalid index-pack output"

mr-q17, 2024-10-27:
Install Raspiblitz via Proxmox / Debian 12.7.0 amd64

- Fresh Debian install, VM via Proxmox 8.2-2

- First install with:
`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh`

- 2nd try:
`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.11/build_sdcard.sh`
 
- Both ends with:
`fatal: fetch-pack: invalid index-pack output`

[image]

adevoss, 2024-10-29:
Just a guess: Firewall / VPN issue?
I had issues before with VPN and github, not only with raspiblitz. Try installing Raspiblitz without VPN.

geco91, 2024-10-29:
Its an issue that happens sometimes and sometimes not - its related to github server response (and internet stability).

geco91, 2024-10-29:
#4823 will not solve but try to reduce this happening