raspiblitz/raspiblitz #4831: v1.11.3rc6 Public Pool locale

NilByte, 2024-11-03:
The app requires  locale set to en_US.UTF-8 in order to  display, refresh and calculate:

_Hash Rate | Session Best Difficulty | Uptime | Last Seen_

In spite having en_US.UTF-8 set, the update to v1.11.3rc6 gave en_GB.UTF-8

https://github.com/raspiblitz/raspiblitz/issues/4566#issue-2291075258

rootzoll, 2024-11-03:
I just tested on my v1.11.3rc6

```
admin@192.168.178.143:~ ₿ cat /etc/default/locale
#  File generated by update-locale
LANG=en_US.UTF-8
```

So what are we missing?

NilByte, 2024-11-03:
I upgraded from v1.11.3rc4 where I had changed locale to en_US.UTF-8 to get Public Pool working.

After upgrade to v1.11.3rc6, the locale was set back to en_GB.UTF-8 by the upgrade and I had to change it to en_US.UTF-8 and reinstall Public Pool app.

I checked my locale in raspi-config and it is set to All Locales there.

If the upgrade to v1.11.3rc6 was intended to set locale en_US.UTF-8, then it didn't work for me.

rootzoll, 2024-11-03:
Can you post your `locale` output here?
```
admin@192.168.178.143:~ ₿ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

rootzoll, 2024-11-03:
BTW when I look into raspi-config on the same test raspberrypi that shows me `LANG=en_US.UTF-8` on the command line I see "en_GB.UFT-8" as selected as `generated` ... but thats just `generated` - same as `en_US.UTF-8` is seleted at the same time in that list - which is  actvely set.

[image]

And with all the default settings after v1.11.3rc6 install I get with publicpool correct statistics for my locale BitAxe miner:
[image]

NilByte, 2024-11-04:
Flashed a second sd card with v1.11.3rc6 and did a fresh upgrade again.

I get:
```
admin@192.168.1.105:~ ₿ locale
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=
```

And the weird Uptime and Last Seen 6.9 month ago which i beleive is due to the locale not being set to en_US.UTF-8.

[image]

Raspi-config has * on:
[image]
[image]

Will try a simple reboot to see if it changes anything. :)

NilByte, 2024-11-07:
If this is pushed to 1.12.0 release, and users experience the problem on 1.11.3, then a manual fix is updating the locale:

```
admin@192.168.1.105:~ ₿ sudo update-locale LANG=en_US.UTF-8
admin@192.168.1.105:~ ₿ source /etc/default/locale
admin@192.168.1.105:~ ₿ cat /etc/default/locale
#  File generated by update-locale
LANG=en_US.UTF-8
```

Followed by uninstalling and reinstalling Public Pool app.

rootzoll, 2025-05-18:
OK I tested so for upcomming v1.12.0 and it default to `en_US` locale - public RC will be out soon. Setting to final testing.

closed 2025-07-30 by rootzoll