raspiblitz/raspiblitz #4835: errorHDD Fail Cleaning HDD Pi5 Migration from MyNode

mokeybear, 2024-11-07:
**Describe the bug**
Attempting to migrate from MyNode to Raspiblitz Pi5 8GB Ram I got thru and had an error stating errorHDD
Fail Cleaning HDD

**To Reproduce**
Start -> Proceed Thru Fresh Setup -> Kept Blockchain data but cleaned drive. -> Waited for 10-15 min and came back to this. 

**Debug file**

Displaying Info Message: Please start Recovery
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
# PROVISION PROCESS with setupPhase(recovery)
# special setup tasks: CLEAN DRIVE & KEEP BLOCKCHAIN
Deleting everything on HDD/SSD while keeping blockchain ...
# raspi_bootdir(/boot/firmware)
Cleaning Blockchain: bitcoin
Using /dev/urandom for random input.
Wipe mode is secure (38 special passes)
Wiping /mnt/hdd/bitcoin/wallet.db CLEANING HDD FAILED:
missing btrfs package
Please report as issue on the raspiblitz github.
###########################################################
# blitz.datadrive.sh status
# raspi_bootdir(/boot/firmware)
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=0
hddRaspiVersion=''

mokeybear, 2024-11-07:
more info I realized after I posted, Drive is formatted into ext4 I dont know if I can migrate it to btrfs or if the data drive needs to be in btrfs

rootzoll, 2024-11-12:
Which version of mynode where you running before?

mokeybear, 2024-11-12:
mynode_raspi45_0-3-24

I got raspiblitz up and running by shutting down, removing the SD card, formatting with windows disk utility and then reflashing with balena.