raspiblitz/raspiblitz #4838: sshd fails to start on raspiblitz-min-v1.11.3-2024-11-03.img.gz

fanaticalfishing, 2024-11-08:
On initial boot sshd fails to start
2024-11-08T19:41:23.108896+00:00 raspberrypi systemd[1]: systemd-hostnamed.service: Deactivated successfully.
2024-11-08T19:41:23.139033+00:00 raspberrypi _background.scan.sh[786]: updating: /home/admin/config.scripts/internet.sh status local
2024-11-08T19:41:23.156942+00:00 raspberrypi _background.sh[740]: skipping rest of tasks because still in setupPhase(recovery)
2024-11-08T19:41:23.215284+00:00 raspberrypi _bootstrap.sh[4880]: rescue-ssh.target is a disabled or a static unit not running, not starting it.
2024-11-08T19:41:23.297581+00:00 raspberrypi systemd[1]: Starting ssh.service - OpenBSD Secure Shell server...
2024-11-08T19:41:23.323447+00:00 raspberrypi sshd[4940]: sshd: no hostkeys available -- exiting.
2024-11-08T19:41:23.324279+00:00 raspberrypi systemd[1]: ssh.service: Control process exited, code=exited, status=1/FAILURE
2024-11-08T19:41:23.324683+00:00 raspberrypi systemd[1]: ssh.service: Failed with result 'exit-code'.
2024-11-08T19:41:23.324952+00:00 raspberrypi systemd[1]: Failed to start ssh.service - OpenBSD Secure Shell server.
2024-11-08T19:41:23.327700+00:00 raspberrypi _bootstrap.sh[4880]: Could not execute systemctl:  at /usr/bin/deb-systemd-invoke line 145.
2024-11-08T19:41:23.644628+00:00 raspberrypi _background.scan.sh[786]: Skipping deeper system scan - setupPhase(recovery) state(system-init)
2024-11-08T19:41:23.661566+00:00 raspberrypi systemd[1]: ssh.service: Scheduled restart job, restart counter is at 1.
2024-11-08T19:41:23.662289+00:00 raspberrypi systemd[1]: Stopped ssh.service - OpenBSD Secure Shell server.
2024-11-08T19:41:23.695736+00:00 raspberrypi systemd[1]: Starting ssh.service - OpenBSD Secure Shell server...
2024-11-08T19:41:23.717350+00:00 raspberrypi sshd[5000]: sshd: no hostkeys available -- exiting.
2024-11-08T19:41:23.720370+00:00 raspberrypi systemd[1]: ssh.service: Control process exited, code=exited, status=1/FAILURE
2024-11-08T19:41:23.721017+00:00 raspberrypi systemd[1]: ssh.service: Failed with result 'exit-code'.
2024-11-08T19:41:23.721368+00:00 raspberrypi systemd[1]: Failed to start ssh.service - OpenBSD Secure Shell server.
2024-11-08T19:41:23.852154+00:00 raspberrypi systemd[1]: ssh.service: Scheduled restart job, restart counter is at 2.
2024-11-08T19:41:23.852805+00:00 raspberrypi systemd[1]: Stopped ssh.service - OpenBSD Secure Shell server.
2024-11-08T19:41:23.872595+00:00 raspberrypi systemd[1]: Starting ssh.service - OpenBSD Secure Shell server...
2024-11-08T19:41:23.873524+00:00 raspberrypi systemd[1]: Reloading.
2024-11-08T19:41:23.896638+00:00 raspberrypi sshd[5003]: sshd: no hostkeys available -- exiting.
2024-11-08T19:41:24.171118+00:00 raspberrypi _background.sh[740]: skipping rest of tasks because still in setupPhase(recovery)

fanaticalfishing, 2024-11-08:
I get the same error with raspiblitz-fat-v1.11.3-2024-11-03.img.gz
2024-11-08T21:34:17.107877+00:00 raspberrypi _bootstrap.sh[4425]: rescue-ssh.target is a disabled or a static unit not running, not starting it.
2024-11-08T21:34:17.150490+00:00 raspberrypi systemd[1]: Stopping ssh.service - OpenBSD Secure Shell server...
2024-11-08T21:34:17.151904+00:00 raspberrypi systemd[1]: ssh.service: Deactivated successfully.
2024-11-08T21:34:17.152282+00:00 raspberrypi systemd[1]: Stopped ssh.service - OpenBSD Secure Shell server.
2024-11-08T21:34:17.181733+00:00 raspberrypi systemd[1]: Starting ssh.service - OpenBSD Secure Shell server...
2024-11-08T21:34:17.203569+00:00 raspberrypi sshd[4433]: sshd: no hostkeys available -- exiting.
2024-11-08T21:34:17.205134+00:00 raspberrypi systemd[1]: ssh.service: Control process exited, code=exited, status=1/FAILURE
2024-11-08T21:34:17.205346+00:00 raspberrypi systemd[1]: ssh.service: Failed with result 'exit-code'.
2024-11-08T21:34:17.205791+00:00 raspberrypi systemd[1]: Failed to start ssh.service - OpenBSD Secure Shell server.
2024-11-08T21:34:17.208797+00:00 raspberrypi _bootstrap.sh[4425]: Could not execute systemctl:  at /usr/bin/deb-systemd-invoke line 145.
2024-11-08T21:34:17.438898+00:00 raspberrypi systemd[1]: ssh.service: Scheduled restart job, restart counter is at 1.
2024-11-08T21:34:17.439524+00:00 raspberrypi systemd[1]: Stopped ssh.service - OpenBSD Secure Shell server.
2024-11-08T21:34:17.459799+00:00 raspberrypi systemd[1]: Starting ssh.service - OpenBSD Secure Shell server...
2024-11-08T21:34:17.482891+00:00 raspberrypi sshd[4449]: sshd: no hostkeys available -- exiting.
2024-11-08T21:34:17.484679+00:00 raspberrypi systemd[1]: ssh.service: Control process exited, code=exited, status=1/FAILURE
2024-11-08T21:34:17.485065+00:00 raspberrypi systemd[1]: ssh.service: Failed with result 'exit-code'.
2024-11-08T21:34:17.485533+00:00 raspberrypi systemd[1]: Failed to start ssh.service - OpenBSD Secure Shell server.
2024-11-08T21:34:17.689902+00:00 raspberrypi systemd[1]: ssh.service: Scheduled restart job, restart counter is at 2.
2024-11-08T21:34:17.693045+00:00 raspberrypi systemd[1]: Stopped ssh.service - OpenBSD Secure Shell server.
2024-11-08T21:34:17.719504+00:00 raspberrypi _background.sh[769]: skipping rest of tasks because still in setupPhase(boot)
2024-11-08T21:34:17.719951+00:00 raspberrypi systemd[1]: Starting ssh.service - OpenBSD Secure Shell server...
2024-11-08T21:34:17.723257+00:00 raspberrypi systemd[1]: Reloading.
2024-11-08T21:34:17.741778+00:00 raspberrypi sshd[4455]: sshd: no hostkeys available -- exiting.

rootzoll, 2024-11-12:
Since you can still cal temrinal, what does starting sshd manually says:
`sudo /usr/sbin/sshd -d`

fanaticalfishing, 2024-11-13:
> Since you can still cal temrinal, what does starting sshd manually says: `sudo /usr/sbin/sshd -d`

Unfortunately I don't have terminal access. I got the above log from reading the sdcard logs with my laptop

fanaticalfishing, 2024-11-15:
It is really strange. I built the sd image with build_sdcard.sh, when I rebooted it the ssh daemon did not start

fanaticalfishing, 2024-11-20:
It is working fine now. There was something wrong with the disk that caused sshd to crash. I copied the blockchain over to a new pi and new disk. it is working fine now.

closed 2024-11-20 by fanaticalfishing