raspiblitz/raspiblitz #4839: RTL for CLN did not work.

neversleep404, 2024-11-09:
Updated from 1.11.1 to 1.11.3 first with MINIMAL SD Card Image then with FATPACK. Same Error with RTL. CPU Load is 5.51.
What could I do?

[image]
[image]

neversleep404, 2024-11-09:
With a disabled RTL, the load is at 2.71

rootzoll, 2024-11-11:
I tested an update/recover with CLN & RTL on a RP4 .. and I was not able to reproduce this so far.

aquilafedele, 2024-11-11:
I have the same error.
Updated from 1.11.2 to 1.11.3, removed RTL, reinstalled RTL and got the same error.
Let me know how I can help providing more details!

neversleep404, 2024-11-11:
It is not so important for me. I have issues since last year. I think I make a payout of my Satoshi and wait for Version 1.12 to start with a clean wallet and a new installation.

neversleep404, 2024-11-11:
without RTL it works perfect

rootzoll, 2024-11-12:
> I have the same error.
> Updated from 1.11.2 to 1.11.3, removed RTL, reinstalled RTL and got the same error.
> Let me know how I can help providing more details!

@openoms can you take a loook?

Hagbard89, 2024-12-11:
same issue here, let me know if you need further informations

rootzoll, 2024-12-12:
@Hagbard89 can you post the result of:
```
sudo netstat -tuln | grep 7000
```

And check if its diffenetent when you deinstall RTL for CLN?

Hagbard89, 2024-12-12:
tcp6       0      0 :::7000                 :::*                    LISTEN
with CRTL deinstalled there's no output

openoms, 2024-12-12:
Downgraded CLN , works with that too.
[image]
Will try a fresh install next.

openoms, 2024-12-12:
you can try removing RTL entirely with the command:
```
config,scripts/bonus.rtl.sh off cl mainnet purge 
```
then reinstall it.

@Hagbard89 please post the exact error you are having aither in the logs or on screen.

Hagbard89, 2024-12-13:
this fixed it! CRTL is now showing all values correctly, no errors during uninstall and reinstall. Thank you very much!

closed 2024-12-17 by rootzoll